MrChromebox/firmware #254: Stuck in UEFI firmware

TheBlayze, 2021-02-16:
So, I updated the UEFI firmware and I am not able to revert back to chrome os. I am just stuck in the firmware of UEFI. I don't know what to do. I tried downloading gallium os and I transfered it over to flash drive and it would not appear as a boot method. If anyone knows what to do please help me. My chromebook is Asus c200

MrChromebox, 2021-02-16:
you're not being very clear as to what you did, or what the issue actually is. if you flashed the UEFI firmware from ChromeOS, then you simply need to boot via USB and install an OS. If your USB isn't showing up under the Boot Menu, then use another and/or try the suggestions on my FAQ page

TheBlayze, 2021-02-16:
Can you please explain to me how you would boot via USB and download an OS. I got the problem by installing the UEFI on chrome OS, wanting to get Linux on my chromebook.

MrChromebox, 2021-02-16:
you need to download a Linux ISO, then write to to USB properly. My FAQ tells you how to do this.

If your USB drive isn't showing up under the Boot Menu, has nothing to do with what's on it or how you wrote it, it's a hardware compatibility issue

TheBlayze, 2021-02-16:
Thanks Mr chromebox for helping me out.
I was able to solve the problem.
I did not have a pc available to write the iso to the flash drive but I had my phone and an otg cable to transfer files to a flash drive. So, I found an app on my phone to write the iso file to the flash drive. And the rest is history

MrChromebox, 2021-02-16:
great, glad you got it sorted :)

closed 2021-02-16 by MrChromebox