MrChromebox/firmware #255: UEFI firmware on stumpy device only detecting maximum 8GB RAM

ssng, 2021-02-18:
I have a Samsung Chromebox series 3 stumpy device. I flashed UEFI firmware on it and installed Ubuntu Linux 20.04. 

I tried to upgrade the RAM but it would only detect up to a maximum of 8GB. I have 2 sticks of 4GB and 2 sticks of 8GB RAMs. No matter what combination of RAM sticks and slots I tried (4+4, 8+8, 4+8, 8+4), it would always report total of 8GB. My RAM sticks are same brand/model so they should not conflict. I have also tested them on a laptop and confirmed that they are all working fine.

Is 8GB the absolute maximum amount of RAM that can be supported by stumpy device?

Thanks BTW for your excellent work in this project.

MrChromebox, 2021-02-18:
I'm running 2x8GB in my STUMPY and no issues with detection. Could be the module density that's an issue, unsure. Would need to check the firmware boot log and see if any clues there

ssng, 2021-02-18:
This is my firmware boot log up to the memory-detection section (in 8+8 config):

```
coreboot-4.12-90-g50c3937717-MrChromebox-4.12 Thu Jun  4 05:17:41 UTC 2020 bootblock starting (log level: 7)...
FMAP: Found "FLASH" version 1.1 at 0x600000.
FMAP: base = 0xff800000 size = 0x800000 #areas = 5
FMAP: area COREBOOT found @ 600200 (2096640 bytes)
CBFS: Locating 'fallback/romstage'
CBFS: Found @ offset 80 size 8a74
BS: bootblock times (exec / console): total (unknown) / 0 ms

coreboot-4.12-90-g50c3937717-MrChromebox-4.12 Thu Jun  4 05:17:41 UTC 2020 romstage starting (log level: 7)...
SMBus controller enabled
Setting up static northbridge registers... done
Initializing Graphics...
Back from systemagent_early_init()
Intel ME early init
Intel ME firmware is ready
ME: Requested 8MB UMA
Starting UEFI PEI System Agent
Read scrambler seed    0x00007581 from CMOS 0x98
Read S3 scrambler seed 0x0000a317 from CMOS 0x9c
FMAP: area RW_MRC_CACHE found @ 5b0000 (65536 bytes)
prepare_mrc_cache: at 0xffdb0040, size ba8
FMAP: area COREBOOT found @ 600200 (2096640 bytes)
CBFS: Locating 'mrc.bin'
CBFS: Found @ offset 19fdc0 size 2fc94
System Agent: Starting up...
System Agent: Initializing PCH
System Agent: Initializing PCH (SMBUS)
System Agent: Initializing PCH (USB)
System Agent: Initializing PCH (SA Init)
SA PciExpress skipped (pcie_init is 0)
System Agent: Initializing PCH (Me UMA)
System Agent: Initializing Memory
System Agent: Done.
System Agent Version 1.2.2 Build 0
ME: Sending Init Done with status: 0, UMA base: 0x1ff8
ME: Requested BIOS Action: Continue to boot
ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : Bring up
ME: Current Operation Mode  : Normal
ME: Error Code              : No Error
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : 0x50
memcfg DDR3 ref clock 133 MHz
memcfg DDR3 clock 1330 MHz
memcfg channel assignment: A: 0, B  1, C  2
memcfg channel[0] config (00620010):
   ECC inactive
   enhanced interleave mode on
   rank interleave on
   DIMMA 4096 MB width x8 dual rank, selected
   DIMMB 0 MB width x8 single rank
memcfg channel[1] config (00620010):
   ECC inactive
   enhanced interleave mode on
   rank interleave on
   DIMMA 4096 MB width x8 dual rank, selected
   DIMMB 0 MB width x8 single rank
MRC_VAR pool occupied [feff7000,feffb92c]
CBMEM:
IMD: root @ 0xacfff000 254 entries.
IMD: root @ 0xacffec00 62 entries.

```

MrChromebox, 2021-02-19:
log shows the DIMMs being detected as 4GB ea, so compatibility issue with the modules. Not something I can fix in firmware as the RAM init is a precompiled blob from Intel

rnizametdinov, 2021-05-30:
> This is my firmware boot log up to the memory-detection section (in 8+8 config):
How did you get  this boot log?

MrChromebox, 2021-05-30:
> How did you get this boot log?

using the coreboot cbmem utility

https://mrchromebox.tech/files/util/cbmem.tar.gz

closed 2021-10-08 by MrChromebox