MrChromebox/firmware #256: Documentation incorrect on AKALI360 for flashing bios

guizmoau, 2021-02-26:
Hello

Thanks a lot for your great great great works on this @MrChromebox !!

I have an ACER Chromebook SPIN 13 AKALI360 and I did install correctly the FULL UEFI Firmware but the only thing is that I did order a suzyQ cable when I actually didnt needed it !!

So please update your compatibility chart for AKALI360 maybe as it require to open and take off battery in order to be able to flash firmware, the suzyQ cable does not work on it as it is not recognized...

I was thinking if someone else in the future as the same model as me (acer chromebook spin13 / AKALI360), it can be reassure that it doesn't need a suzyQ cable to flash the bios...only battery removal suffice

MrChromebox, 2021-02-26:
it's not incorrect, AKALI/AKALI360 both use CR50, and all CR50 devices can have the WP temp disabled by booting with the battery disconnected -- as detailed on https://wiki.mrchroembox.tech/Firmware_Write_Protect

guizmoau, 2021-02-26:
hum i see better in details now, I was missing that bit 

(...)
CR50 models (2017+): On all Chromebook models with the CR50/Google Security Chip (all Kabylake/Apollolake and newer models), the !WP pin is controlled by the CR50. Under factory configuration, the CR50 sets the WP state to follow the battery sense line, **so disconnecting the battery will disable the hardware write protect**. It's also possible to change the WP state using the closed-case debugging (CCD) features of the CR50, along with a special USB-C debug cable. See the CCD section below.
(...)

I see so the suzyQ debug cable is a "cosmetic" option so you don't have to open your laptop right ?

closed 2021-02-26 by guizmoau