MrChromebox/firmware #257: Samsung Chromebox model XE300M22, enable WOL or Wake on AC

ShapeShifter499, 2021-03-03:
I remember asking sometime ago on a IRC channel and back then nothing was figured out. I'm wondering if by any chance any new information might have been found out. 

The goal is to enable some sort of Wake-on-LAN or Wake-on-AC functionality so this Samsung Chromebox I have could be used as a server and I'd have a way to reliably turn it back on if I'm away from the machine in the event of a power outage. 

If such support couldn't be enabled in firmware/bios is there some external solution anyone might have to reliably turn on a computer when Wake-on-LAN or Wake-on-AC isn't supported?

MrChromebox, 2021-03-03:
wake-on-AC is programmable in firmware, needs the southbridge and SIO configured properly. Would require custom firmware build.

Don't believe device is WOL capable

ShapeShifter499, 2021-03-03:
> wake-on-AC is programmable in firmware, needs the southbridge and SIO configured properly. Would require custom firmware build.

How hard would it be to generate such firmware?

MrChromebox, 2021-03-03:
generate? not too long. test and ensure it works? significantly longer, especially since I recall something broke since the last time I tried this on a Haswell Chromebox

ShapeShifter499, 2022-12-23:
Any possible updates to this issue?

ShapeShifter499, 2022-12-23:
I no longer require this as I have today got a workaround, thanks to some kind folks at ##electronics chat over on irc.libra.chat 

After walking me though some testing with a multimeter. We established that the system puts 3.3v through the power button and looks for a drop in voltage when the button shunts the power to 0v when pressed. 

Someone had the idea to put a capacitor that was between 10uF and 100uF across the pins of the button. When the system powers up on AC the capacitor draws current to charge up, causing a dip in voltage just long enough to appear to the rest of the system as a power button press. 

Thus a really dead simple analog hardwire workaround to essentially make the computer "wake on AC"

Some pictures of the mod. My soldering job is a little janky due to clipping the capacitor lead a little short and needing to put in a wire. 

Before 
[image]

After 
[image]

closed 2022-12-23 by ShapeShifter499