MrChromebox/firmware #260: Lenovo N21 not turning on after flashing firmware

RemiAQ, 2021-03-26:
I installed the custom firmware from the website and followed all of the instructions, and after telling it to reboot it didnt power back on. The backlight isnt turning on, and the power button isnt doing anything. However my theory is that it just, didnt reboot and turned the screen off and disabled the keyboard. Because theres still a bit of heat coming from it. Any ideas on what to do here? I really dont want to ship it off to mrchrome to get it unbricked, but worse comes to worse im willing to pay the money.

MrChromebox, 2021-03-27:
try hooking up to an external display, GIMMER is known to have issues with the internal display init

RemiAQ, 2021-03-28:
> 
> 
> try hooking up to an external display, GIMMER is known to have issues with the internal display init

Nothing is coming up on my ext display. Too late now I guess, already ordered a new chromebook :\

closed 2021-03-28 by RemiAQ

MrChromebox, 2021-03-29:
why order a new Chromebook vs a USB flasher to unbrick it?

MrChromebox, 2025-09-07:
@mattygjak just reflash the UEFI firmware using the USB programmer. Ensure the device is fully charged as it should still power on even if the firmware is corrupted.