MrChromebox/firmware #261: WCOM Digitizer was not registered in Caroline's device tree

waitlamp, 2021-03-28:
hello,
in https://github.com/MrChromebox/firmware/issues/216 , we found the reason why s-pen not work after upgrade firmware.

now I might found the reason why Wacom W9013 digitizer was incorrectly disabled

Chromebook Pro ( Caroline ) has a Wacom W9013 EMR digitizer,
It is not the only Chromebook device has digitizer.

DELL 7486 , ACER Spin13 (Nami). Also used Wacom EMR digitizer Chip.

If a Chromebook has digitizer, 
there are some Digitizer configurations in variants overridetree.cb or devicetree.cb,

a part of spin13(Nami)'s devicetree.cb like:
(I'm not sure what the "PENH" is)
```
end # I2C #1
		device pci 15.2 on
			chip drivers/i2c/hid
				register "generic.hid" = ""WCOM005C""
				register "generic.desc" = ""WCOM Digitizer""
				register "generic.irq" = "ACPI_IRQ_LEVEL_LOW(GPP_D1_IRQ)"
				register "generic.probed" = "1"
				register "generic.reset_gpio" = "ACPI_GPIO_OUTPUT_ACTIVE_LOW(GPP_D3)"
				register "generic.reset_delay_ms" = "20"
				register "generic.has_power_resource" = "1"
				register "generic.disable_gpio_export_in_crs" = "1"
				register "generic.wake" = "GPE0_DW2_01"
				register "hid_desc_reg_offset" = "0x1"
				device i2c 0x9 on end
			end
			chip drivers/generic/gpio_keys
				register "name" = ""PENH""
				register "gpio" = "ACPI_GPIO_INPUT_ACTIVE_HIGH(GPP_E8)"
				register "key.dev_name" = ""INST""
				register "key.linux_code" = "SW_PEN_INSERTED"
				register "key.linux_input_type" = "EV_SW"
				register "key.label" = ""pen_insert""
				register "key.wakeup_route" = "WAKEUP_ROUTE_DISABLED"
				device generic 0 on end
			end
		end # I2C #2
```
but  WCOM Digitizer was not registered in Caroline overridetree.cb
this is the entire I2C part of Caroline overridetree.cb
```
device domain 0 on
		device pci 15.0 on
			chip drivers/i2c/generic
				register "hid" = ""ATML0001""
				register "desc" = ""Atmel Touchscreen""
				register "irq" = "ACPI_IRQ_EDGE_LOW(GPP_E7_IRQ)"
				device i2c 4b on end
			end
		end # I2C #0
		device pci 15.1 on
			chip drivers/i2c/generic
				register "hid" = ""ATML0000""
				register "desc" = ""Atmel Touchpad""
				register "irq" = "ACPI_IRQ_LEVEL_LOW(GPP_B3_IRQ)"
				register "wake" = "GPE0_DW0_05"
				device i2c 4a on end
			end
		end # I2C #1
		device pci 19.2 on
			chip drivers/i2c/nau8825
				register "irq" = "ACPI_IRQ_LEVEL_LOW(GPP_F10_IRQ)"
				register "jkdet_enable" = "1"
				register "jkdet_pull_enable" = "0"	# R389
				register "jkdet_polarity" = "1"		# ActiveLow
				register "vref_impedance" = "2"		# 125kOhm
				register "micbias_voltage" = "6"	# 2.754
				register "sar_threshold_num" = "4"
				register "sar_threshold[0]" = "0x0c"
				register "sar_threshold[1]" = "0x1c"
				register "sar_threshold[2]" = "0x38"
				register "sar_threshold[3]" = "0x60"
				register "sar_hysteresis" = "1"
				register "sar_voltage" = "0"		# VDDA
				register "sar_compare_time" = "0"	# 500ns
				register "sar_sampling_time" = "0"	# 2us
				register "short_key_debounce" = "2"	# 100ms
				register "jack_insert_debounce" = "4"	# 64ms
				register "jack_eject_debounce" = "4"	# 64ms
				device i2c 1a on end
			end
			chip drivers/i2c/generic
				register "hid" = ""INT343B""
				register "desc" = ""SSM4567 Left Speaker Amp""
				register "uid" = "0"
				device i2c 34 on end
			end
			chip drivers/i2c/generic
				register "hid" = ""INT343B""
				register "desc" = ""SSM4567 Right Speaker Amp""
				register "uid" = "1"
				device i2c 35 on end
			end
		end # I2C #4
		device pci 1e.6 on  end # SDCard
	end
end
```

I guess it's the reason why wcom digitizer was recognized as ACPI0C50 not WCOM0C50 (I'm not sure if this model correct) both on Linux and windows.

and I think that's why developer think there was no device on I2C2 and incorrectly disabled it.

you can compare [Caroline](https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/Samsung%20Chromebook%20Pro%20XE510C24-Caroline/sysinfo.UwjeIiKZUM/devtree.txt) 's wacom-hid-descriptors file with another [samsung laptop](https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/Samsung%20930QAA/sysinfo.lflVw3erhl/devtree.txt) has similar wacom EMR Digitizer (chips model are different)
caroline:
`/sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-3/i2c-ACPI0C50:00/0018:2D1F:0163.0001
`
930QAA
`/sys/devices/pci0000:00/0000:00:15.2/i2c_designware.1/i2c-2/i2c-WCOM0032:00/0018:2D1F:002E.0002`

thanks!

MrChromebox, 2021-03-29:
> and explained why developer think there was no device on I2C2 then incorrectly disabled it.

I'm the developer :) It was simply my error when converting to the overridetree format, since I didn't look at the ACPI to see that I2C2 was used.

I'm fairly certain that everything is correct as per the stock firmware for CARLOLINE. If you feel it would benefit from an additional ACPI device being declared, that's something we can test

waitlamp, 2021-03-29:
> I'm the developer :)

sorry, don't get me wrong, I'm not familiar with firmware development.

In short words, wacom Digitizer on Caroline now can be correctly recognized as ACPI pen input devices, and can be driven by system general-purpose driver. but Manufacturer ID ( WCOM ) was not recognized due to not being registered in firmware.

if the declaration in firmware can make the manufacturer ID to be correctly identified.
This may facilitate system driver development/install and has no side effects.

It does so in the newer chromebook devices' firmware.
So I think you can also test it on caroline.
This might improve the firmware

Thanks！

MrChromebox, 2021-03-29:
well, first thing would be to figure out the correct WCOM* HID string to use. I can easily change the HID/CID, but not sure that using NAMI's string is correct since they aren't the same model

waitlamp, 2021-03-30:
> well, first thing would be to figure out the correct WCOM* HID string to use. I can easily change the HID/CID, but not sure that using NAMI's string is correct since they aren't the same model

The good news is it seems wacom driver is insensitive to HID string `WCOM****` both on Linux and Windows.
Some people even tried [remapping HID](http://www.forum.wacom.eu/viewtopic.php?f=11&t=4545) string to make device can install driver.

I also found two devices used same W9013 EMR digitizer chip
Galaxybook 12:`WCOM0028`
Zbook X2:`WCOM002E`
It seems wacom uses different HID string though chips are same. So may need further work.
another thing is, both of them was not contained in wacom official tablet driver (HP offered an OEM Driver)
There is a [PID lists](http://www.wacomeng.com/component/0519931883/supportedPIDs.txt) listed supported devices

Additionally, I heard that windows ink general driver will limit devices pressure level to 1024 level
but w9013 has being confirmed working as a 4096 level digitizer.

now I'm planning to remapping HID string and check if my device can install feel it driver.
I'll comment again if I get success.

MrChromebox, 2021-03-30:
it's easy enough for me to change the string in the test build if you want, just LMK

MrChromebox, 2021-04-26:
should be fixed in 4.13 release -- please confirm

waitlamp, 2021-06-19:
> should be fixed in 4.13 release -- please confirm

digitizer work well in new firmware release , and indeed with WCOM005C string

And I found that It seems only wacom windows driver use the string to identify if driver can be installed
their Linux driver or other opensource driver don't care about this string.

I tried to modified ACPI in Windows test mode without modified the firmware
, used a string contained in their driver `.inf`file

although wacom windows driver can be loaded in device manager after remapping ACPI
control panel can't dectect the device.

linux driver has better compatibility
linuxwacom added `.tablet` for caroline
after that caroline's tablet can be detected in GNOME
with new firmware, it can be detected as usual.

So I think the new firmware has no compatibility issues with caroline

thanks!

closed 2021-06-19 by waitlamp