MrChromebox/firmware #262: Careena UEFI issue

logan2k, 2021-03-28:
On Careena with latest Mrchromebox UEFI firmware:
Windows 10 x64 20H2 got BSOD during booting USB, with message "ACPI_BIOS_ERROR".
On xUbuntu 20.10, the touchpad do not work(maybe a driver issue instead of UEFI).

MrChromebox, 2021-03-29:
> Windows 10 x64 20H2 got BSOD during booting USB, with message "ACPI_BIOS_ERROR".

the script told you this before flashing, not sure how much more clear I can make it that it's not capable of running Windows (and I have no interest or time to try and resolve it).

> On xUbuntu 20.10, the touchpad do not work(maybe a driver issue instead of UEFI).

no idea, it's a standard Elan touchpad, worked for all my testers so I'd assume it's something specific to Ubuntu 20.10

logan2k, 2021-03-29:
Thank you.
I cannot see the warnings because I cannot use the script due to network behind a proxy. I have to manually flash using "flashrom" directly.
So are all Windows versions incompatible with this device?
Have you tested any Linux distros that have all functions working?

MrChromebox, 2021-03-29:
Windows isn't incompatible with your device, your device (or more precisely, its firmware) isn't compatible with Windows. Even if you could boot Windows, there are no drivers for many of the components. It's a complete waste of time.

I've not tested Linux lately but suspect it will at least boot there. I can't keep track of what works/doesn't for 100+ Chromebooks. A community Wiki would be useful here

closed 2021-03-31 by logan2k