MrChromebox/firmware #263: Non-ChromeOS Devices board names

bam80, 2021-03-29:
Hi,
in https://mrchromebox.tech/#devices table of supported devices,
these column are not filled for  Non-ChromeOS Devices:
-  Board Name
-  WP Method
Particular interesting are Purism Librem Mini devices. Do they have this information?
Also, Purism Librem Mini v2 has no RW_LEGACY Firmware nor UEFI Firmware (Full ROM) columns filled, what does it mean?
Thank you.

MrChromebox, 2021-03-29:
these aren't "issues." 

Non-ChromeOS devices don't have a ChromeOS board name. Nor do they have hardware write protection that must be disabled. 

RW_LEGACY firmware exists only in the context of stock Google ChromeOS firmware, so wouldn't apply to non-ChromeOS devices. 

I haven't added support for the Mini v2 yet.

closed 2021-03-29 by MrChromebox

bam80, 2021-03-29:
OK, but could we add to the table EGLOBAL devices which are OEM for Purism and seem to be the same hardware?
https://www.aliexpress.com/i/32850708878.html
If so, how to identify they are the same hw other than via the board names?

MrChromebox, 2021-03-29:
Eglobal is not Purism's ODM. They also use the same model number for multiple platforms in the same chassis so wouldn't make sense. You could flash Purism's firmware (or your own built from upstream coreboot) onto one of these but obviously Purism isn't going to facilitate that

bam80, 2021-03-29:
> They also use the same model number for multiple platforms in the same chassis so wouldn't make sense. You could flash Purism's firmware (or your own built from upstream coreboot) onto one of these

So how could I know if I can flash coreboot to the particular model e.g. on the link above?

MrChromebox, 2021-03-29:
short answer, you don't. You can check the platform and compare to the Purism models, and flash the one that matches, but no guarantee it will work. 

I have no idea how those Eglobal models compare to Purism's, or what differences may exist. I do know ours come with an EC firmware update (from our ODM) that those are likely to not have, and will likely result in the cooling fan to not work under coreboot.

bam80, 2021-03-29:
Thank you for the concrete information.
Just a few more questions:
> I do know ours come with an EC firmware update (from our ODM) that those are likely to not have

- Does it mean the EC firmware can't be updated even on Librem mini boards?
- Can your ODM be disclosed?
- Do Librem laptops come from some ODM too (meaning they are not Purism's domestic design as Librem 5)?

> no guarantee it will work

- Is there some recovery procedure? (for Librem's)

> I haven't added support for the Mini v2 yet.

- Does it mean Librem mini v2 comes with proprietary firmware?
- What is the point to flash firmware from here if other Librem's come with coreboot already?

Overall, I appreciate the value of porting to coreboot and consider to purchase the device from Purism, too.
Just wanted to consider all the possibilities. I think it would be honest toward the community. 
One of the options I'm considering is donating to you directly.
Thank you.

MrChromebox, 2021-03-29:
> Does it mean the EC firmware can't be updated even on Librem mini boards?

Purism doesn't provide a mechanism to do so, nor do they provide EC updates to non-Purism customers (the EC update is applied by our ODM before we receive the boxes).

> Can your ODM be disclosed?

not my place to do so, nor do I have that information. Try emailing Purism support?

> Do Librem laptops come from some ODM too (meaning they are not Purism's domestic design as Librem 5)?

they are built to our specification. The base (unmodified) board design is not Purism's, it's a reference design from the ODM (which I assume is based on an Intel reference design)

> Is there some recovery procedure? (for Librem's)

flashrom an an external programmer (eg, ch341a)

> Does it mean Librem mini v2 comes with proprietary firmware?

not at all, why would it? Purism provides official, OSS-based firmware for all Librems

> What is the point to flash firmware from here if other Librem's come with coreboot already?

my firmware uses Tianocore (UEFI) as the payload, unlike the official Purism firmware which uses SeaBIOS or Heads. This is simply an alternative to the official offering, just like with Chromebooks.

bam80, 2021-04-03:
> the EC on the Minis is an ITE IT8528E and can be updated via ITE's Windows or EFI update utilities

Thanks.
Do Minis have Free EC Firmware? I don't see IT8528E support here: https://github.com/system76/ec/blob/master/doc/controllers.md, not sure if it's the right place to look, though.
If not how one could find the firmware image Minis have and make sure it supports the cooling fan under coreboot?

MrChromebox, 2021-04-05:
> Do Minis have Free EC Firmware? 

no

> I don't see IT8528E support here: https://github.com/system76/ec/blob/master/doc/controllers.md, not sure if it's the right place to look, though.

you would want to look in Purism's fork of S76's EC firmware. We've added support for the IT8528E for the Librem 14, so support for the Mini could in theory be added, but I don't see that happening.

> If not how one could find the firmware image Minis have and make sure it supports the cooling fan under coreboot?

you can't. it's not publicly available, and Purism doesn't have the ability to distribute/provide it.

bam80, 2021-04-05:
I see, thank you for the insight. From now on, I think Purism's forum would be better place to discuss this.

bam80, 2021-04-11:
> I haven't added support for the Mini v2 yet.

Is it something being working on?
I need to decide if it's worth to buy most powerful device now and then wait a bit for a firmware update or it's better to choose a lower spec device with better support. UEFI is essential for me. Any advise?

MrChromebox, 2021-04-12:
> Is it something being working on?
> I need to decide if it's worth to buy most powerful device now and then wait a bit for a firmware update or it's better to choose a  lower spec device with better support. UEFI is essential for me. Any advise?

the only reason the Mini v2 isn't supported is because it wasn't released when I did my last coreboot release (2020.06.04). It will be included in the next release (whenever I have time) along with the Librem 14