MrChromebox/firmware #266: Firmware request: Lenovo C13/Morphius

pettter, 2021-04-09:
My understanding is that there's no support for this device yet - Is there something me as an owner of one can help with speeding up or enabling development of?

I'd like to get away from ChromeOS as fast as humanly possible.

MrChromebox, 2021-04-09:
> I'd like to get away from ChromeOS as fast as humanly possible.

then you bought the wrong device.

AMD Chromebooks have always run native Linux poorly. These new Ryzen Chromebooks are even worse, since the PSP is even more omnipresent and harder to work around. I have a Morphius Chromebook and can't even get it to boot. I don't plan to spend days/weeks on it either -- seems the PSP requires running verified boot or won't release the x86 cores out of reset.

pettter, 2021-04-10:
> then you bought the wrong device.

Seems like it, indeed.

Thanks for the quick response. 

Feel free to close this, unless you want it up For Posterity - I'll be returning my device.

MrChromebox, 2021-04-10:
sorry to be the bearer of bad news. Chromebooks with 10th-gen Intel chips look to fare a bit better w/r/t Linux support, and I'll be releasing UEFI firmware for them soon

pettter, 2021-04-10:
No worries mate, thanks for all that you do!

MrChromebox, 2021-05-05:
ok I have a test build if you still have the device and want to play around :)

pl-luk, 2021-05-19:
Hi,
I have bought a C13 as well (unfortunately before I stumbled upon this issue :/) but I am willing to spend a few weeks on figuring this out. Do you still have the test build?

MrChromebox, 2021-05-19:
>  I am willing to spend a few weeks on figuring this out.

figuring what out exactly?

pl-luk, 2021-05-19:
Figuring out how to get tianocore up and running

MrChromebox, 2021-05-19:
there's nothing to figure out, it works

pl-luk, 2021-05-19:
> I have a Morphius Chromebook and can't even get it to boot.
I am sorry I have misunderstood that one.

So if I have not messed something up again... Coreboot + tianocore is working but Linux is not booting?

MrChromebox, 2021-05-19:
> So if I have not messed something up again... Coreboot + tianocore is working but Linux is not booting?

linux boots and runs fine. I'll probably add support with the upcoming 4.14 release

pl-luk, 2021-05-19:
Okay, thank you :)

MrChromebox, 2021-09-03:
closing as MORPHIUS now has working RW_LEGACY/AltFw

closed 2021-09-03 by MrChromebox

remigasas, 2021-09-11:
Glad to hear MORPHIUS has working RW_LEGACY, thank you for adding support.
Is there any chance to get "(UEFI) Full ROM" working ?

MrChromebox, 2021-09-11:
It's working just fine, but mainline Linux support for AMD mobile platforms is still lacking. 99% of the time people flash the UEFI Full ROM, find out things aren't working, and go back to stock, and somehow in those 3 hours manage to lose their stock firmware backup. It's just a headache I don't want to deal with. If you know what you're getting into and want to flash a UEFI Full ROM, email me (mrchromebox at gmail) and I'll build one and provide a link

remigasas, 2021-09-14:
It is perfectly understandable, that it is not fun to deal with user related issues and consult them.
I would like to take your offer for custom build, but I think it is important to update regularly.

Hagb, 2023-06-14:
I tried to use the UEFI Full ROM for Morphius on my C13, and found that it can boot grub but grub can not boot Linux.
After spending days on coreboot, I just found that setting "Option ROM execution type" to "Secure mode" (`CONFIG_PCI_OPTION_ROM_RUN_YABEL=y`) in coreboot configuration solves the problem. With this configuration, grub booted by coreboot can boot Linux<5.19 (same as RW_LEGACY Tianocore).
The following is a working configuration (modified from [configs/zen2/config.morphius.uefi](https://github.com/MrChromebox/coreboot/blob/MrChromebox-4.20/configs/zen2/config.morphius.uefi)):
```
CONFIG_USE_AMD_BLOBS=y
CONFIG_VENDOR_GOOGLE=y
CONFIG_NO_POST=y
# CONFIG_CONSOLE_SERIAL is not set
CONFIG_VBOOT=y
CONFIG_BOARD_GOOGLE_MORPHIUS=y
CONFIG_PCI_OPTION_ROM_RUN_YABEL=y
CONFIG_FRAMEBUFFER_SET_VESA_MODE=y
CONFIG_FRAMEBUFFER_VESA_MODE_USER=y
CONFIG_FRAMEBUFFER_VESA_MODE=0x1d4
CONFIG_EDK2_TAG_OR_REV="origin/upp_202301"
```
(PS: `origin/upp_202304` version of edk2 doesn't work.)

Hope this works on other ryzen chromebook.

Hagb, 2023-06-14:
I don't know why Linux>=5.19 fail to boot (both kernels of Debian and Ubuntu). Booting them causes an immediate reboot without any output. I also tried to turn off iommu (`iommu=off`, refer to [this page](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1990068)) and/or use serial (AP console when using SuzyQ cable, usually `/dev/ttyUSB1`) to get the output from kernel (`earlycon=uart,mmio32,0xfedc9000 console=ttyS0,115200`), but get nothing. (in kernel<5.19 the output can be got)

MrChromebox, 2023-06-14:
I'm not bothering with any Zork devices until we have a fixed PSP bootloader, it's pointless to try and work around the issues currently. Use RW_LEGACY on Zork for now.

Hagb, 2023-06-15:
> I'm not bothering with any Zork devices until we have a fixed PSP bootloader, it's pointless to try and work around the issues currently. Use RW_LEGACY on Zork for now.

I have used the the coreboot built with the above configuration for two days. It seems working well just like RW_LEGACY—so far I haven't met any problem that RW_LEGACY don't have. (RW_LEGACY also fail to boot Linux>=5.19.)