MrChromebox/firmware #268: Acer C740: pch_wildcat_point-virtual-0 Unusually High Temperatures

jasonmerc, 2021-04-13:
I'm running Firmware 4.12 from MrChromebox's repo for my Acer C740.  Totally stock chromebook hardware wise, other than a 128GB SSD I put into it.  For some reason, when running "sensors" on my Kubuntu 20.04 install it shows "pch_wildcat_point-virtual-0" running at an extremely high temperature.  Everything else is normal, 40-50C.  This wildcat thing though can go above 100C easy.

Here is copied output from the "watch sensors" command:

_**coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +44.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +43.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +43.0°C  (high = +105.0°C, crit = +105.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          11.54 V
curr1:       447.00 mA

pch_wildcat_point-virtual-0
Adapter: Virtual device
temp1:        +95.0°C

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +42.8°C  (crit = +104.0°C)**_

I don't understand what could be causing this.  No part of the chromebook feels hot.  I even applied fresh thermal paste to the heatsink.  If any other logs are needed please let me know

MrChromebox, 2021-04-13:
I'd guess it's a kernel bug, otherwise a coreboot bug. Either way it's nothing to worry about

dwillmore, 2023-01-18:
FWIW, same here with an ASUS Chromebox 2 (CN62).  It reports 90+ degrees all the time.  Moves a little with system activity, but not much.  Ohter temps look okay.

closed 2026-11-04 by MrChromebox