MrChromebox/firmware #270: USB bakcup device failed to mount; cannot proceed.

rkumar679, 2021-04-26:
Hi Team

Just having a issue with the ChromeOS Firmware utility script v2021-04-23. When it comes to the stage to backup stock firmware, the USB is not able to mount.
Get Error 'USB backup device failed to mount; cannot proceed'.
I am running this script on a Lenovo x131e chromebook. Also i have tried with  3-4 different USB sticks still does not work.

please help. thanks.
[image]
[image]

MrChromebox, 2021-04-26:
just fixed the USB backup issue.

But also seems your device is not being correctly identified, given the Device and Platform sections of the menu are incorrect. Not sure what's going on there but need to sort it out before it will flash the correct image for your device.

what's the output of: 
`sudo crossystem hwid`
and
`sudo dmidecode -s system-product-name`

please

rkumar679, 2021-04-26:
thanks a lot. I will try again.

sudo crossystem hwid = X86 STOUT TEST 8762

sudo dmidecode -s system-product-name = Stout

MrChromebox, 2021-04-26:
ok, board identification should be fixed now -- please try again

closed 2021-04-26 by MrChromebox

rkumar679, 2021-04-27:
thanks all working now. great support.

Breakfastcrafty, 2023-01-31:
its happening to me.
HWID: QUAWKS C2A-C6P-Q7Y

dmidecode: Quawks

yuftheyuf, 2023-10-13:
please i have the same problem ,can u help

MrChromebox, 2023-10-13:
> please i have the same problem ,can u help

use a FAT32-formatted USB stick

thamnis, 2024-01-18:
_I've the same problem and this is my outputs_

> thanks a lot. I will try again.
> 
> sudo crossystem hwid = PHASER360 D4J-B3K-C42-K4N-Y2I-22C-I8Z
> 
> sudo dmidecode -s system-product-name = Phaser

ThallaNikhil, 2024-09-02:
Hi team 
I am facing issue in my hp crombook, when I was installing IOS my pen drive was not detecting 
Showing that 
USB backup device failed to mount; cannot proceed 
[image]

ACEU8X, 2025-03-08:
> Hi Team
>
> Just having a issue with the ChromeOS Firmware utility script v2021-04-23. When it comes to the stage to backup stock firmware, the USB is not able to mount.
> Get Error 'USB backup device [...]

I got the same issue but this is very new to me and I am confused what you guys were talking about in that post. Can you help please I'm in the same situation let me know how to fix it

AkiraVI, 2025-03-14:
> [@ACEU8X](https://github.com/ACEU8X) is the USB FAT32 formatted? with a single partition?

I have same issue Lenovo Flex 5i Chromebook 14

[image]

prootserver, 2025-07-29:
For anyone else reporting this issue ("USB backup device failed to mount; cannot proceed. Ensure your USB is FAT32-formatted and try again."), here's how I was personally able to resolve this.

Launch elevated command prompt
>diskpart
>list disk
>select disk  # (number of intended device, matching output of above command)
>clean (triple check that you've selected the right disk first!)

And then opening diskmgmt.msc, choosing to initialize the disk as MBR (NOT GPT), and formatting as FAT32.