MrChromebox/firmware #271: Acer C740 Chromebook: Black screen after UEFI Firmware Install

malandles, 2021-04-26:
Same behaviour as #90 and #194, although not the same device.

After installing the firmware, did a reboot and inserted a new boot USB device (Linux distro), but the screen stayed black.

Tried without the boot drive and with an HDMI attached, no progress.

Any suggestions?

MrChromebox, 2021-04-26:
try a hard reset (Refresh+Power).  If no luck, then https://wiki.mrchromebox.tech/Unbricking

malandles, 2021-04-26:
Will try unbricking, thank you

closed 2021-04-26 by malandles

Alphix, 2021-04-27:
Version 20210423 also didn't work for me (black screen, no luck). Reverting to 20200604 did work.

Alphix, 2021-04-27:
Opened a separate issue #272

malandles, 2021-04-27:
Thank you, will follow in your steps!

malandles, 2021-09-20:
Unbricked with `https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-auron_paine-mrchromebox_20210725.rom`, thanks!