MrChromebox/firmware #272: Acer C740 (auron paine): refuses to power on after installation of latest UEFI Firmware (previous version works)

Alphix, 2021-04-27:
Tried updating from SeaBIOS to UEFI full ROM firmware on my Acer C740 (auron paine).

Used the mrchromebox.tech/firmware-util.sh script, which installed:
https://mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-auron_paine-mrchromebox_20210423.rom

Set my existing Debian installation up for UEFI boot, rebooted. Black screen (no backlight, but the power LED stayed on).

Got a CH341a USB flash programmer, flashed the same rom again (after checking the sha1, also did a verification run of flashrom after the firmware was flashed). Reassembled computer, crossed fingers, pressed power button.

The C740 remained powered on for 0.5 seconds, then powered down (that is, the power LED turned on for 0.5 secs, then turned off).

Flashed an earlier version:
https://mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-auron_paine-mrchromebox_20200604.rom

And everything worked like a charm :)

MrChromebox, 2021-04-27:
fixed build posted

Alphix, 2021-04-28:
Thanks, I tried updating to https://www.mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-auron_paine-mrchromebox_20210427.rom this morning using the `firmware-util.sh` script.

After rebooting, the screen lit up and showed the coreboot logo together with the "Press ESC..." message (so that's an improvement from `20210423`).

Problem is, I can't boot any EFI images. Tried selecting various images (`systemd-bootx64.efi` and various Linux kernels prepared using `sicherboot` and sucessfully booted using using the `20200604` version) from the coreboot menu, but every time the screen just goes black and nothing happens (backlight stays on, power LED stays on).

Alphix, 2021-08-16:
@MrChromebox A friendly ping...

MrChromebox, 2021-08-16:
@Alphix have you tested the 2021.07.25 release? I fixed PAINE long ago

Alphix, 2021-08-16:
@MrChromebox thanks, will try it out soonish (in about two weeks)

Alphix, 2021-09-03:
@MrChromebox : just tested the 2021.07.25 release and it works like a charm, thank you :)

closed 2021-09-03 by Alphix

malandles, 2021-09-20:
Unbricked my C740 with `2021.07.25`, thanks!

NIKT1313, 2022-09-23:
Any chance to GET rom 2021.07.25 .????? link or something else? After update to last version (2022..07.18) on C740 i have black screeen and nothing else :(