MrChromebox/firmware #273: Samsung Chromebook 3 Starts Booting when Plugged In and Shut Down

techguy112, 2021-05-03:
My Samsung Chromebook 3 with the Mrchromebox UEFI Software starts to try and turn on and boot when the computer is plugged in and shut down. It is kind of annoying, sometimes I ask someone else to plug in my computer while I am doing something and they do but the computer starts booting and because it is running it does not charge fast enough at certain times. It did not do this until the latest coreboot update. Can you try and fix this?

MrChromebox, 2021-05-03:
this is standard Chrome EC behavior and isn't related to the most recent coreboot/UEFI firmware

techguy112, 2021-05-03:
Ok, thanks for the fast response!

foxsouns, 2021-05-17:
> My Samsung Chromebook 3 with the Mrchromebox UEFI Software

Okay, I know this is off-topic, but @techguy112, I have a question.
How? How do you have a functioning UEFI boot CELES? Mine bricked immediately on flash, and has been dead for months. Could you post your specific model code? Mine always seemed different from the other numbers I was seeing with the CELES; I wonder if some were sent out with the absolute worst NVRAM handling and general firmware possible, and some were normal. Haha, anyways rant over. RIP, you 5 year old brick.

MrChromebox, 2021-05-17:
@foxsouns just unbrick per my instructions - https://wiki.mrchromebox.tech/Unbricking

closed 2021-05-17 by MrChromebox