MrChromebox/firmware #274: Boot menu shortcut key?

agnivo007, 2021-05-05:
Is there a shortcut key for a quick boot menu (not the Esc key shown at boot which opens to the main menu) which only lists the boot sources available for booting and lets the user choose, just like other mainstream BIOS (F9=HP,F12=Dell) ?

My system is a Lenovo x131e (stout) chromebook flashed to latest 4.13 MrChromebox full UEFI firmware.

MrChromebox, 2021-05-05:
there is no shortcut key for the boot menu

agnivo007, 2021-05-06:
Is there any way (in coreboot code) that it can be implemented in future versions?

MrChromebox, 2021-05-06:
it's not a function of coreboot, but of Tianocore (the UEFI payload). 

Could it be implemented? sure, it shouldn't be too too hard. but I can't commit to doing so

agnivo007, 2021-05-06:
OK, I'll look forward to this. The reason I'm asking for it is, I need such a setup to instruct my child to press F(N) to select Windows if he chooses to boot it instead of Cloudready. I don't want him to fiddle with the full-bios menu brought up by pressing ESC. He might inadvertently change settings or delete boot entries there. So, a simple F(N) or Q (easier for chromebooks as they don't have FN keys marked) for Quick Boot Menu would be a great addition. I guess not only me, but many more would appreciate the new feature.

closed 2026-11-04 by MrChromebox