MrChromebox/firmware #276: Tidus - ThinkCentre Chromebox - second m.2 PCIe not detecting specific device?

kirk444, 2021-05-10:
I am using the full UEFI on a ThinkCentre Chromebox (Tidus). I have what I believe is a firmware compatability issue, though I am not certain.
I am trying to add an I/O Crest Ethernet adapter ( http://www.iocrest.com/index.php?id=2178 ) to one of the slots, but it is never detected.

The board has two PCIe m.2 slots (one normally houses the WiFi). I've confirmed that the wifi is detected in either slot (and that if I populate both slots with Intel 7260 wifi cards they're both detected). The ethernet NIC, however, is not recognized in either slot. I took a somewhat similar Lenovo M710q and replaced the Wi-Fi with the ethernet, and it works fine there (confirming that the NIC itself is functional).

I'm not sure what additional information might be helpful to troubleshoot this, but happy to provide anything that may help.

MrChromebox, 2021-05-11:
boot Linux with the adapter in the slot, then use the coreboot cbmem utility to pull the firmware boot log. Download https://mrchromebox.tech/files/util/cbmem.tar.gz, extract, then run `sudo ./cbmem -1 > cbmem.log` and throw the log on pastebin. Do the same with `sudo dmesg > dmesg.log` and post the pastebin links here

kirk444, 2021-05-11:
It's on the motherboard, between the wifi slot and the cpu, here's a not great picture, but you can still see it: https://www.ebay.com/itm/223823187600

MrChromebox, 2021-05-11:
does the WiFi card work in the 2nd socket?

MrChromebox, 2021-05-11:
I have a mini PC here with 2x RT8168 so not an issue.

can you put the adapter in the other PC in which it was detected and then do `sudo lspci -vv -s y.00.0` where y is the PCI bus number for the NIC

kirk444, 2021-05-11:
Don't have access to that other one at the moment, I'll find something to put it in and get the requested output though. In the meantime I might have bricked the Tidus somehow (it no longer displays any video, and doesn't seem to connect to the network after boot) so it might be moot, anyways.

kirk444, 2021-05-12:
Here's the output from lspci from another host: https://pastebin.com/kUAJZjnU

Also, good news, not bricked, just doesn't want to output over displayport anymore, using HDMI for now.

ericdespain, 2021-07-07:
On Tidus, the m.2 memory card slot only supports SATA.  A SATA express card will work in it, but a pcie card will not.  I tried both.
FYI,
Eric

kirk444, 2021-07-07:
I've confirmed that a PCIe wifi card works fine in that slot, and also that a SATA SSD does not, I do not believe that your observations are correct. "SATA Express" is another connector type, and is different from M.2.

closed 2026-11-04 by MrChromebox