MrChromebox/firmware #278: 4.13 booting issues compared to 4.12, Sandybridge

jd7654, 2021-05-17:
Have a few Acer C710 units. Was regenerating one and decided to try new firmware 4.13 upgrade from previous 4.12.

Problem 1: Windows 10 boot USB failed to boot, error about need to verify it is 64 bit, which it was.
Narrowed problem down to this: the USB was made MBR partition scheme, as it was made to work for both BIOS and UEFI computers, which it did work on 4.12. Made a new Windows 10 USB with GPT partitions, and then it booted in 4.13.

Other USB drives worked OK like Brunch, Cloudready, or GalliumOS.

Problem 2: rEFInd boot manager cannot directly boot 64-bit Linux kernels, which worked fine with 4.12. No error, computer just reboots. I can boot these Linux kernels (like antiX) by putting them in the Grub2 menu of another Linux like Gallium OS. But I lost the functionality of booting directly from rEFInd main boot screen. Have to hop through another Grub2 boot first.

MrChromebox, 2021-05-29:
this will be fixed in 4.14

jd7654, 2021-06-27:
Noticed same issue on Haswell, Acer C720P with 4.13.
In both cases, downgraded to 4.12 fixed issues.
Will wait for 4.14 fix before any full rom upgrades.

closed 2021-10-08 by MrChromebox