MrChromebox/firmware #279: Is there a mirror of older firmware? Want to obtaining last full firmware with UEFI and Legacy boot via SeaBIOS.

tinkeros, 2021-05-22:
I cloned scripts and did:
git grep monroe $(git rev-list --all) | grep rom

Only 2 of the 27 rom images are downloadable (coreboot_tiano-monroe-mrchromebox_20200604.rom and coreboot_tiano-monroe-mrchromebox_20210423.rom).

I would very much like to obtain the last full firmware which has both UEFI boot and Legacy boot via SeaBIOS for monroe.  Is there a way to do this or is my only option to try to figure out the necessary git commits and try to build it myself?

Thank you for any help you may be able to offer!

MrChromebox, 2021-05-22:
> Only 2 of the 27 rom images are downloadable

yes, keeping hundreds of old firmware images around takes up a lot of space

> I would very much like to obtain the last full firmware which has both UEFI boot and Legacy boot via SeaBIOS for monroe. 

there's never been a single firmware that did both. Back when I built separate images, the legacy boot ones weren't even tested. The value of those older images is pretty much nil.

>  Is there a way to do this or is my only option to try to figure out the necessary git commits and try to build it myself?

you can build both from upstream coreboot without issue. Just need to extract a few blobs from your existing or stock firmware.

tinkeros, 2021-05-22:
Okay, thank you for the response!  I was looking at your website and saw some references which said "removed Legacy Boot (SeaBIOS) Full ROM firmware" which made my think maybe there was firmware with both at some point.  I will try to extract the blobs and build a legacy boot firmware to experiment with.  I've got a BIOS flasher so I can fix it if I mess it up.

closed 2021-05-22 by tinkeros

MrChromebox, 2021-05-22:
MONROE has an ethernet jack that gets its MAC address from the firmware, so if you care about that, might need to build from my tree and persist the RO_VPD region like my script does.

blob wise, really just need mrc.bin. The IFD and ME firmware are only needed if flashing the entire image externally, and should be extracted from a clean firmware image (not from a running system).