MrChromebox/firmware #280: Lenovo N20P cannot init panel after full UEFI firmware install

hexdump0815, 2021-05-26:
hello,

i just tried to install you latest uefi firmware (as of today, so i guess it is the coreboot 4.13 release from 2021.04.21) on a lenovo n20p clapper (4gb ram/16gb emmc version) and after the reboot the displays backlight turns on, but not the display. luckily i found the hint to use hdmi and that works, so i'm happy to not having to reflash it the hard way by hand (which i once did successfully some years ago on an acer chromebook, but i'm happy to not having to do it again).

what kind of information do you need to be able to diagnose and maybe fix the problem? i can boot a xubuntu 20.04 installer iso with a live system easily and collect information, which might be helpful for you.

best wishes and looking forward to you response - hexdump

hexdump0815, 2021-05-26:
some more info: according to the drm edid info the panel seems to be an InfoVision M116NWR1 R7 - here is a full dump of the edid:
```
00000000  00 ff ff ff ff ff ff 00  26 cf 89 04 00 00 00 00  |........&.......|
00000010  00 17 01 04 80 1a 0e 78  0a 12 30 91 56 53 92 28  |.......x..0.VS.(|
00000020  1e 50 54 00 00 00 01 01  01 01 01 01 01 01 01 01  |.PT.............|
00000030  01 01 01 01 01 01 d5 1c  56 a6 50 00 23 30 28 20  |........V.P.#0( |
00000040  3c 00 00 90 10 00 00 19  39 13 56 a6 50 00 23 30  |<.......9.V.P.#0|
00000050  28 20 3c 00 00 90 10 00  00 19 00 00 00 fe 00 49  |( <............I|
00000060  6e 66 6f 56 69 73 69 6f  6e 0a 20 20 00 00 00 fe  |nfoVision.  ....|
00000070  00 4d 31 31 36 4e 57 52  31 20 52 37 20 0a 00 17  |.M116NWR1 R7 ...|
```
the panel is working fine if a hdmi monitor is attached and i can use it with linux just fine - only coreboot does not seem to be able to show anything on it without a monitor connected to hdmi.

MrChromebox, 2021-05-26:
the Tianocore GOP driver is failing to init the display, no idea why. The only other option is to have coreboot run the VBIOS, but then you're stuck with 1024x768 resolution, and there's issues with suspend/resume. No good option for Baytrail unfortunately. And it works fine here on SWANKY so I can't reproduce

hexdump0815, 2021-05-26:
would it make sense to try the june version from 2020? i have that one working well on an acer cb3 111 which is baytrail too. is it possible to flash an older than latest version of the firmware with your scripts out of the box?

in case i would install linux on the chromebook as it is, will it boot then well without hdmi connected and i'll get a screen in linux then (which i guess is iitializing the panel on its own) - so in the end i would just loose coreboot menu etc. access (which i can get with an external monitor in case i need it) and everything else will work normally - is this assumption correct?

a lot of thanks by the way for keeping all those bioses around to make those old and sometimes no longer supported chrombooks into some still very useable devices ....

MrChromebox, 2021-05-26:
> would it make sense to try the june version from 2020? i have that one working well on an acer cb3 111 which is baytrail too. is it possible to flash an older than latest version of the firmware with your scripts out of the box? 

The script doesn't provide the ability to flash older versions unfortunately. The 2020.06.04 release uses the same display init method, so no difference would be expected.

I'd need to create a debug build of Tianocore that logs to the coreboot log in memory, and hope it catches whatever the issue is. But that's a bit of effort

hexdump0815, 2021-05-27:
there is no pressure at all - if you have anything to test (and which is not risking to brick the chromebook :) ) just let me know ...

meanwhile i have installed linux on it and besides not being able to access the coreboot bios menu and grub without using an external monitor it is working fine so far - maybe it might be worth to mention the hint to test with an external monitor in your unbricking doc, so that people at least give it a try before learning how to recover their chromebook the hard way :)

MrChromebox, 2021-05-29:
drop me an email MrChromebox at gmail and I'll provide a test build to flash with Tianocore debug logging that might help determine the issue

hexdump0815, 2022-03-10:
i just reinstalled the latest uefi firmware for clapper (from july 2021) as for some strange reason the firmware you created back then for me did only see 2gb of my 4gb ram and i can confirm now that this latest firmware is working well with proper display support and 4gb ram ... time to close this issue :)

thanks a lot for all your efforts - hexdump

closed 2022-03-10 by hexdump0815