MrChromebox/firmware #282: ASUS CN60 (Panther, Haswell, Full ROM / UEFI) won't boot from USB, believe after MrChromebox-4.13 (04/23/2021) upgrade

TrueBuilder, 2021-06-04:
My issue seems similar to how "Acer C740 (auron paine): refuses to power on after installation of latest UEFI Firmware (previous version works) #272" _ends_ (not starts).  Here's the evidence:
[1]  I believe this started after I upgraded to the latest Haswell firmware; note this shows it, but isn't the screen where I actually upgraded:
> ChromeOS Device Firmware Utility Script [2021-04-27] 
>  (c) Mr Chromebox <mrchromebox@gmail.com> 
> *********************************************************
> **   Device: ASUS Chromebox CN60 (PANTHER)
> ** Platform: Intel Haswell
> **  Fw Type: Full ROM / UEFI
> **   Fw Ver: MrChromebox-4.13 (04/23/2021)
> **    Fw WP: Disabled
> *********************************************************
> ** [WP] 1) Install/Update UEFI (Full ROM) Firmware 
> ** [WP] 2) Restore Stock Firmware 
> **      C) Clear UEFI NVRAM 
> *********************************************************
> Select a menu option or
> R to reboot  P to poweroff  Q to quit

[2]  I've so far experimented with Pop!_OS and PureOS, sha256summed everything I could, and built USBs with Gnome Disks (restore image), with dd, and with (Balena) Etcher; all fail (so I'm currently dual-booting GalliumOS "inside" (eMMC?), and Pop!_OS on an antique HDD.  FWIW, in the years prior to "MrChromebox-4.13 (04/23/2021)" I've installed GalliumOS, openSUSE, Pop!_OS, and probably other Linux distros I'm not remembering.

[3]  For some installs I pull the internal M2 SSD, and so only have install USB & target storage, for others I skip pulling internal.  All seem to fail the same way(s?).

[4]  I believe that there was an initial error different, before I began suspecting the firmware, but I didn't of course record that.  What I can reproduce now looks like this (notice the single space after "3.0" -- I'm not sure that it's exactly one space before "failed" or "verify"):
> Booting from 'USB: Kingston DataTraveler 3.0 ' failed: verify it contains a 64-bit UEFI OS.
> Press any key to continue booting...

[5]  I've executed multiple "[WP] 1) Install/Update UEFI (Full ROM) Firmware" and "C) Clear UEFI NVRAM" pairs, sometimes one first, sometimes the other, but that doesn't seem to help.  So everything's running now, dual-boot, but I'm stuck as regards future installs, only hope that the June release of Pop!_OS 21.04 does upgrade and doesn't break anything.

Respectfully,
TrueBuilder

MrChromebox, 2021-06-04:
@TrueBuilder please drop me an email (MrChromebox at gmail) and I'll give you a test build to flash, should resolve issue

MrChromebox, 2021-06-28:
closing as this appears to be an issue with user's hardware, not reproducible

closed 2021-06-28 by MrChromebox

Hurricos, 2021-11-30:
> closing as this appears to be an issue with user's hardware, not reproducible

I can reaffirm your closure. I have had success with booting Panther from USB this month, both from a USB 3.0 drive (uas-based), and through a USB-2.0-to-SATA adapter. It must have just been this one user's hardware.

TheForgottenKing, 2022-12-13:
I can reproduce this issue on a CN60 using the most recent firmware revision.

MrChromebox, 2022-12-13:
@TheForgottenKing what issue exactly? what firmware version?

the simple fact is that not every USB stick is going to be compatible with the firmware, because USB specs are unclear and hard to implement properly. I have about a dozen different USB sticks here, diff manufacturers and capacities, and all are correctly detected on my CN60.

TheForgottenKing, 2022-12-13:
That was quick. Disregard, I hardware flashed the newest revision (20221129) with a CH341A and now it works. I may have been on a slightly older version before.

MrChromebox, 2022-12-13:
4.18.0/.1 did have some USB enumeration/detection issues; 4.18.2 should be damn close to 100% working