MrChromebox/firmware #283: How to come back to 4.12 version

Spok974, 2021-06-11:


MrChromebox, 2021-06-11:
no way to install older versions, keeping track of one version for each of the ~100 device supported is hard enough. Not sure what's different but I know others have installed MacOS without issue on the 4.13 firmware.

Spok974, 2021-06-13:
Thanks for the answers ! I know a lot of people who have the same issue like me ... if you have a good tutorial or tricks, I could try to make a YouTube video for help the others if I can do it ! Thanks

MrChromebox, 2021-06-13:
I have no idea, I don't test against MacOS. Check the chrultrabook subreddit or other Hachintoshing forums

Spok974, 2021-06-17:
Thank you 
Every tutorial I find on the web are under 4.12 version and its work great like my older hackintosh but with 4.13 impossible to do it actually 

It is possible to copy a bios with 4.12 firmware with your online script? If I find somebody who has the same chromebox can I downgrade 4.13 to 4.12 with a copy of the bios?

I hope you anderstand why I mean 😄
Thanks you so much bro

MrChromebox, 2021-06-18:
no, my script doesn't provide option to flash older versions, and if you use the firmware from another device you will run into issues with the serial number and LAN MAC address being copied

TheRandMan, 2021-07-07:
> I know others have installed MacOS without issue on the 4.13 firmware

@MrChromebox, do you happen to have a source you can point me to for this? Everything I've found points to the same issue with 4.13 and MacOS (kernel panics on reboots/startup and during installation). My best guess is still that it's NVRAM related but doesn't seem to be specific to a model. Maybe generation specific though (Broadwell).

MrChromebox, 2021-07-07:
I don't, I don't use MacOS. I have no way of testing 100+ devices (or even 10+ platforms) for each release.

TheRandMan, 2021-07-08:
Yeah, I get that. Was just hoping you could point me in the direction of somebody who has had success with MacOS and 4.13 so I could compare notes.

toanlm1981, 2021-08-18:
@TheRandMan did you try 4.14 version?

Spok974, 2021-08-29:
Waldoxhm said: bigsur version of efi is updated. I've fully tested with coreboot 4.14, I got kernel panic when boot up, it may due to the bugfix of IGPU and the way nvram stores changed. I managed to downgrade bios to 4.12 check https://github.com/waldoxhm/scripts. 1.burn a linux image to a sdcard with balenaEtcher. 2.boot cn62 to this livecd os. 3.follow the guide run the command.

MrChromebox, 2021-08-30:
@Spok974 that repo is a bad idea, there's no guarantee I'll keep those older versions around. It also won't maintain the Ethernet MAC address on Chromeboxes since the script logic isn't set up to handle going back to the older VPD format

waldoxhm, 2021-09-26:
@TheRandMan simplely set <code>Booter->Quirks->DisableVariableWrite</code> and <code>NVRAM->LegacyEnable</code> to <code>true</code> in config.plist will do the trick in my case. And use emulated NVRAM with LogoutHook.

TheRandMan, 2021-09-28:
> @TheRandMan simplely set <code>Booter->Quirks->DisableVariableWrite</code> and <code>NVRAM->LegacyEnable</code> to <code>true</code> in config.plist will do the trick in my case. And use emulated NVRAM with LogoutHook. 

Thought I tried this but will try again. Thanks.

closed 2026-11-04 by MrChromebox