MrChromebox/firmware #284: 3gb of ram out of 8

helprequester4394, 2021-06-22:
The bios detects that I have 8gb of ram, but when I get in to linux (64 bit) it only detects 3gb of ram and says "paravirtualized kernel" in the initialization. I tried placing mem=8g in grub, but am not sure if this is an issue with firmware or linux. Linux also detects the two soldered on sodimm sticks as 4gb each in lshw. But when I use free I do not get 8gb of ram instead I get 2800 and so free MB of RAM. 
This is on thinkpad c13 yoga, and i have not yet removed wp because the cable not being in stock.
I am trying different linuxes to see if it is a kernel issue or if it is an issue with the firmware. I am not sure which but I think it is the legacy rw firmware which I used to boot the usb.

MrChromebox, 2021-06-22:
so this is on MORPHIUS running the RW_LEGACY firmware?

helprequester4394, 2021-06-22:
> so this is on MORPHIUS running the RW_LEGACY firmware?

Yes. Until I get a suzyqable I cannot set any gbb flags. But it is strange because it is listed fine in bios on coreboot. Once linux boots it does not show more than 3gb. I am trying different distros but it is problem on all of them.

MrChromebox, 2021-06-23:
setting GBB flags won't change anything, nor will trying different distros -- it's a bug in the build of Tianocore I used for the RW_LEGACY payload. I'll update with a fixed build tomorrow

MrChromebox, 2021-06-23:
RWL update pushed - update and LMK if issue resolved

helprequester4394, 2021-06-24:
I installed and ran it. Now there is 8gb of ram available. But now Vesa does not work and i can not get x to work. 
`fb_add_framebuffer_info_ex: channel bit mask=24 and BPP=32 don't match`
`This is a driver bug`
I can get curse framebuffers to work. Like splash graphic for debian. But if I boot in to live I have color garbage on the screen and the mouse responds. Keyboard does not. Tested on Void Debian Ubuntu Alpine and Parabola.
I blacklisted amdgpu and tried using vesa and ati but neither work. And neither does trying to run without the gpu driver. Nothing appears on external monitor.

MrChromebox, 2021-06-24:
<shurgs> these AMD Chromebooks are just a nightmare to run Linux natively on, even moreso on top of the stock Google firmware

helprequester4394, 2021-06-24:
Found fix. I had to blacklist the amdgpu driver in grub and not in mkinitcpio or dracut. Now it runs well.

MrChromebox, 2021-06-24:
so we can close this, yes?

helprequester4394, 2021-06-25:
Yes

closed 2021-06-25 by MrChromebox