MrChromebox/firmware #287: Add firmware option to disable powering on upon lid opening

delipunch, 2021-07-09:
This a feature request to implement an option in the full UEFI firmware menu to enable or disable the device powering on when you open the lid. It should default to the current behavior, enabled, so as to not disrupt the expected funtionality of the firmware, which you were concerned with [here](https://github.com/MrChromebox/firmware/issues/151#issuecomment-507951920).

MrChromebox, 2021-07-14:
I can't see taking the time to implement this, since it would need to pass the option state between Tianocore, coreboot, and the EC. And there's no current mechanism to do so.

delipunch, 2021-07-14:
So when you say that there's no mechanism to do this, you're saying the EC can't read from NVRAM currently? I suppose that's the issue closed, as you have little time to implement this.
Maybe consider an extra build per board, to be selected at install time, but to be fair you already have one build to maintain for >95% of chromeos boards, and 2 or 3 for I'd guesstimate >50% of boards.

closed 2021-07-14 by delipunch

MrChromebox, 2021-07-16:
ChromeEC uses its own battery-backed RAM, it doesn't share NVRAM with coreboot or Tianocore. All 3 have different non-volatile storage regions/methods