MrChromebox/firmware #288: Windows 10 installation media not identified as a viable OS

JJDelisle, 2021-07-12:
Reproduce the issue:
- Install Windows 10 boot media onto USB
- insert the USB into the device with the relevant firmware
- Set USB as primary boot device
- Screen displays text stating "Booting from 'USB: ---' failed; verify it contains a 64-bit UEFI OS."

This happened across 3 different flash drives. I tried booting from the boot file (EFI>boot>bootx64.efi) in the USB and it brought me to the windows installation and only ran into trouble after it recognizing that the SSD does not contain enough storage and Windows 10 doesn't allow you to install to an SD

MrChromebox, 2021-07-14:
Sure sounds to me like the USB media was created improperly. How did you "install" the Windows 10 ISO to USB? Using what app? what settings? what ISO file?

SSD storage requirements or install target limitations are not a firmware issue of any kind

MrChromebox, 2021-09-03:
closing due to inactivity and that this isn't a firmware issue

closed 2021-09-03 by MrChromebox

AmalgamGemini, 2024-01-30:
Open this ticket back up please. MrChromebox. I am having the same issue, and since you asked him of the details, I shall provide you with mine.

1. Not Win10, is, SuperNano Win7 64x - created using an option on the program e.g. Power ISO, option = Create a Bootable USB drive.
2. The Drive I'm attempting to place this file onto are in the following details.

Laptop/Chromebook; Lenovo N21 Model Name; 80MG - Mfg; 2015/04/28
Medium OS to be installed on default HDD: Internal - 16Gig 
RAM: 4Gig

Please; instruct me on how to proceed with the issue at hand. 

Thank you Sir.

CP

MrChromebox, 2024-01-30:
@AmalgamGemini if the issue you're having is the same as the OP -- you're getting the error "Booting from 'USB: ---' failed; verify it contains a 64-bit UEFI OS" -- then that's exactly the issue: your boot media isn't bootable on a 64-bit UEFI system. on top of that, there is zero support in terms of firmware or drivers for anything earlier than Windows 10. And all Windows support is handled by coolstar / on chrultrabook.com.