MrChromebox/firmware #289: Link (Chromebook Pixel 2013) slow Graphics

renr4, 2021-07-15:
Along with the [No turbo](https://github.com/MrChromebox/firmware/issues/195) issue on Link, I also noticed that the HW acceleration works slow. For example `vblank_mode=0 glxgears` gives me around 6000 fps on the stock firmware, but on UEFI it's 3000. So, it's 2 times slower. And the whole UI (scrolling pages, moving windows) slightly lagging. I tried different distros and kernes (even ubuntu 14.04), and no changes. This issue is not on the OS side.

MrChromebox, 2021-07-16:
tried Cloudready or Brunch for most accurate comparison? could be a Chromium kernel tuning issue. Or directly related to lack of turbo boost since that effects the GPU too IIRC

renr4, 2021-07-16:
No, but frame drops were noticeable without tests. I tried both legacy and UEFI on the same distro, with the same kernel params. UEFI were about 2x slower.

renr4, 2021-09-14:
The recent update didn't fix it. Same slow graphics and no turbo.

MrChromebox, 2021-09-15:
not surprising as no specific changes were made to address the issue

bpvarsity, 2022-02-24:
Can confirm the issue with my Chromebook Pixel 2013 as well

elektroencefalografista, 2022-07-03:
Same on mine, running the latest firmware

elektroencefalografista, 2022-08-06:
It seems like the issue is related to CPU power being misreported.
[image]
In hwinfo on Windows the CPU power is being reported around 35-36W with very little change compared to CPU load, while PL1 limit is set at 17W. 
This would definitely explain no turbo and poor GPU performance (due to clocks being stuck at the lowest state). 
@MrChromebox is there a chance you could take another look at this?

MrChromebox, 2022-08-07:
interesting, I'll see what I can find

MrChromebox, 2022-08-07:
looks like the non-turbo ratio is locked at 18 (1.8GHz) preventing SpeedStep (or any sort of power management) from functioning. Not seeing the issue on a SandyBridge Chromebox so need to figure out what's different

elektroencefalografista, 2022-08-07:
That sounds correct to me, iirc this CPU has a non-turbo ratio of 18 and turbo of 28 for single thread workloads.
I don't think the issue is related to Speedstep. I've used [this tool](https://github.com/horshack-dpreview/setPL) to compare the chromebook to my Thinkpad X220 (SB-based) and both behave the same way - if CPU package power is higher than PL1, the ratio won't go over non-turbo clocks. Both laptops turbo properly if PL1/PL2 are set sufficiently high.
I still think the CPU power is being misreported for some reason. I saw ~47-49W being reported at 2.6GHz with all threads loaded. Sounds unreasonable for an U-series Ivy Bridge CPU.

MrChromebox, 2022-08-08:
I agree, ~33W pkg power makes no sense. But also seems like the non-turbo multiplier is fixed at 18, which it shouldn't be. I'm unable to find another U-series SNB/IVB model with coreboot to compare; the x220/230 use M-series CPUs (I tested an x230 here and it does not display the same behavior w/r/t the multiplier or pkg power, so same as your x220)

coolstar, 2023-06-30:
Turbo boost is working fine here on Link w/ coreboot 4.20 and Win10 22H2

renr4, 2023-09-06:
Just installed UEFI on my link and both [turbo boost](https://github.com/MrChromebox/firmware/issues/195) and GPU seem to be working fine now.

closed 2023-09-06 by MrChromebox