MrChromebox/firmware #290: Acer c740 usb boot

myruz, 2021-07-21:
[image]

I tried to install linuxmint but stuck where it says the OS is UEFI 64-bits

What is the problem and how to manage it?
Thanks

MrChromebox, 2021-07-21:
is the error message somehow unclear? You need to be using a 64-bit ISO. It needs to be written to USB properly. I have no idea if that's what you did because "tried to install linux mint" doesn't actually tell me anything.

myruz, 2021-07-22:
thank you for your reply. yes i'm unclear of the error message.

I all ready used 3 thumbdrive. using rufus for the bootable usb.

I tried linux mint 64bits, even the 32bits and also ubuntu.

still the error 
"booting from usb; kingston 'failed ; verify it contains 64-bits UEFI OS

now i'm stuck here. hoping for a solution..

myruz, 2021-07-22:
Problem solved. I change from bootable iso to bootable DD. anyway..thanks. i solved the problem by checking all the issues here.

MrChromebox, 2021-07-22:
the FAQ page is pretty explicit on using Rufus in dd mode :)

closed 2021-07-22 by MrChromebox