MrChromebox/firmware #292: Pixelbook 2017 Problems After Upgraded To The Latest Firmware

Mtres11, 2021-08-03:
Hi there
After upgraded to the latest firmware(2021.07.25 coreboot 4.14)
My Pixelbook 2017 suddenly caught in a mess.
The keyboard doesn't work any more.
Two Type-C USBs don't work any more. 
Try to Hard-Reset. Failed.
Try to flash back the previous Firmware. Failed. (Because of the usbs doesn't work, I can not even boot into linux)
So what can I do next?

MrChromebox, 2021-08-03:
hmm, others have tested and reported no issues, not sure why you'd be having any problems like this. Not much you can do if kb doesn't work other than let the battery drain/die?

jmontleon, 2021-08-05:
There are two of experiencing problems with suspend using 4.14 on the Pixelbook 2017 using Fedora 34. The other user has been experiencing problems with Windows, but I can't vouch for that. Fedora 34 would suspend and resume fine for me with 4.13.

https://www.reddit.com/r/PixelBook/comments/oqefs1/linux_on_pixelbook_with_working_sound/h7syf4d?utm_source=share&utm_medium=web2x&context=3

MrChromebox, 2021-08-05:
the problem appears to be some users are experiencing a failure with the RW EC firmware upgrade, which causes it to revert to the much older factory RO EC firmware, which has issues with suspend and they keyboard under Windows. I'm trying to figure out the best way to proceed that won't brick user's computers

MrChromebox, 2021-08-06:
I've updated the 4.14 UEFI firmware for the Pixelbook with a fixed version, please update and close issue if solved - thx

Mtres11, 2021-08-07:
Thx for your effort.
All things work fine after upgraded to the latest firmware.
Thx again.

closed 2021-08-07 by Mtres11

jmontleon, 2021-08-07:
Also worked for me. Thank you!