MrChromebox/firmware #293: Enable/Support SecureBoot

nschimme, 2021-08-12:
Right now SecureBoot doesn't seem to be enabled within the builds and it is a mandate for Windows 11. It seems like Tianocore supports SecureBoot already upstream but we haven't enabled it within the builds.

https://github.com/tianocore/tianocore.github.io/wiki/How-to-Enable-Security

MrChromebox, 2021-08-12:
> It seems like Tianocore supports SecureBoot already upstream but we haven't enabled it within the builds.

following that link...

> Note: Information for this section was written based on the UDK2010.SR1 release

wildly out of date would be an understatement. There is no SecureBoot  support in upstream edk2, not for UefiPayloadPkg anyway. First would need to get TPM support sorted out. And I have less than zero time to spend on this currently.

bob808, 2021-08-18:
Secure Boot would be useful without TPM at first., for self signing. If it's easier to roll out that is.

Baitinq, 2022-08-26:
@MrChromebox I think it would be really cool if we enabled secureboot by default, more linux distros are adopting secureboot properly and I think this would be really useful.

MrChromebox, 2022-08-26:
I have no plans to implement this feature myself. If/when it comes to upstream EDK2, my fork will pull it in. I simply don't have the time to do this.

pietrushnic, 2022-09-02:
I'm looking at this from perspective of Dasharo. I would like to create disposable latop from my old CELES. AFAIK Dasharo edk2 fork support UEFI Secure Boot I wonder it just matter of replacing payload or there is more work needed here.

MrChromebox, 2022-09-02:
@pietrushnic if Dasharo's edk2 repo is public, I'm happy to pull in the required patches from there :)

pietrushnic, 2022-09-02:
@MrChromebox yes, it is public. Although I took a look and it I don't see straight forward way to integrate. We have something we call [common base](https://github.com/Dasharo/coreboot/pull/221) for coreboot. It relies on newer coreboot thank your next branch. [Here](https://github.com/Dasharo/coreboot/pull/221/files#diff-ab85ead684a4f0cffcb158801be8d2c2b8610e2cf46ce4ca78e92edbeb2154c9) you can see enable support for iPXE, Secure Boot and HDD pass. That's coreboot part and [EDKII](https://github.com/Dasharo/edk2/tree/dasharo).

pietrushnic, 2022-09-02:
I tried to quickly look at most recent tag in your coreboot tree, but realized that I have no idea what is upstream and what is not and which patches I have to port to make things wokr as expected (without bricking) while building customized version. Merging our stuff which based on newer coreboot would lead to more work than I had time to do. Also I have no idea how to deploy results. Your scripts work great with released firmware, but I assume I would have to hack those to use coreboot.rom based on Dasharo.

BTW my idea for Qubes OS Summit 2022 disposable laptop would be:
- WP enabled
- coreboot+UEFI with Secure Boot
- [safeboot](https://safeboot.dev/)

If I can help any way please let me know, I'm available whole weekend.

pietrushnic, 2022-09-03:
I tried to build `next` but it looks like it has some issues with blobs repo:

```shell
debian@build-engine:/tmp$ git clone https://github.com/MrChromebox/coreboot.git -b next
Cloning into 'coreboot'...
remote: Enumerating objects: 709227, done.
remote: Counting objects: 100% (138380/138380), done.
remote: Compressing objects: 100% (32664/32664), done.
remote: Total 709227 (delta 104326), reused 136050 (delta 102513), pack-reused 570847
Receiving objects: 100% (709227/709227), 172.82 MiB | 10.95 MiB/s, done.
Resolving deltas: 100% (549469/549469), done.
Updating files: 100% (19994/19994), done.
debian@build-engine:/tmp$ cd coreboot/
debian@build-engine:/tmp/coreboot$ git submodule update --init --recursive --checkout
Submodule '3rdparty/amd_blobs' (https://review.coreboot.org/amd_blobs) registered for path '3rdparty/amd_blobs'
Submodule 'arm-trusted-firmware' (https://review.coreboot.org/arm-trusted-firmware.git) registered for path '3rdparty/arm-trusted-firmware'
Submodule '3rdparty/blobs' (https://github.com/MrChromebox/blobs.git) registered for path '3rdparty/blobs'
Submodule '3rdparty/chromeec' (https://review.coreboot.org/chrome-ec.git) registered for path '3rdparty/chromeec'
Submodule '3rdparty/cmocka' (https://review.coreboot.org/cmocka.git) registered for path '3rdparty/cmocka'
Submodule '3rdparty/ffs' (https://review.coreboot.org/ffs.git) registered for path '3rdparty/ffs'
Submodule '3rdparty/fsp' (https://review.coreboot.org/fsp.git) registered for path '3rdparty/fsp'
Submodule 'intel-microcode' (https://review.coreboot.org/intel-microcode.git) registered for path '3rdparty/intel-microcode'
Submodule '3rdparty/intel-sec-tools' (https://review.coreboot.org/9esec-security-tooling.git) registered for path '3rdparty/intel-sec-tools'
Submodule 'libgfxinit' (https://review.coreboot.org/libgfxinit.git) registered for path '3rdparty/libgfxinit'
Submodule 'libhwbase' (https://review.coreboot.org/libhwbase.git) registered for path '3rdparty/libhwbase'
Submodule 'opensbi' (https://review.coreboot.org/opensbi.git) registered for path '3rdparty/opensbi'
Submodule '3rdparty/purism-blobs' (https://source.puri.sm/coreboot/purism-blobs.git) registered for path '3rdparty/purism-blobs'
Submodule '3rdparty/qc_blobs' (https://review.coreboot.org/qc_blobs.git) registered for path '3rdparty/qc_blobs'
Submodule '3rdparty/stm' (https://review.coreboot.org/STM) registered for path '3rdparty/stm'
Submodule 'vboot' (https://review.coreboot.org/vboot.git) registered for path '3rdparty/vboot'
Submodule 'util/nvidia-cbootimage' (https://review.coreboot.org/nvidia-cbootimage.git) registered for path 'util/nvidia/cbootimage'
Cloning into '/tmp/coreboot/3rdparty/amd_blobs'...
Cloning into '/tmp/coreboot/3rdparty/arm-trusted-firmware'...
Cloning into '/tmp/coreboot/3rdparty/blobs'...
Cloning into '/tmp/coreboot/3rdparty/chromeec'...
Cloning into '/tmp/coreboot/3rdparty/cmocka'...
Cloning into '/tmp/coreboot/3rdparty/ffs'...
Cloning into '/tmp/coreboot/3rdparty/fsp'...
Cloning into '/tmp/coreboot/3rdparty/intel-microcode'...
Cloning into '/tmp/coreboot/3rdparty/intel-sec-tools'...
Cloning into '/tmp/coreboot/3rdparty/libgfxinit'...
Cloning into '/tmp/coreboot/3rdparty/libhwbase'...
Cloning into '/tmp/coreboot/3rdparty/opensbi'...
Cloning into '/tmp/coreboot/3rdparty/purism-blobs'...
Cloning into '/tmp/coreboot/3rdparty/qc_blobs'...
Cloning into '/tmp/coreboot/3rdparty/stm'...
Cloning into '/tmp/coreboot/3rdparty/vboot'...
Cloning into '/tmp/coreboot/util/nvidia/cbootimage'...
Submodule path '3rdparty/amd_blobs': checked out '83c44ad892094cad5da7b9562660fdbf4a9dd64c'
Submodule path '3rdparty/arm-trusted-firmware': checked out 'e0a6a512b51558b64eb500e6b731e4c743050af2'
fatal: remote error: upload-pack: not our ref 733f45efc019f555506f730fe8d6a39a2e6b944d
Fetched in submodule path '3rdparty/blobs', but it did not contain 733f45efc019f555506f730fe8d6a39a2e6b944d. Direct fetching of that commit failed.
```

MrChromebox, 2022-09-04:
> I tried to build next but it looks like it has some issues with blobs repo:

fixed

pietrushnic, 2022-09-04:
@MrChromebox thank you. I will test it in couple minutes. During yesterday development I found couple other issue, but it is related to long lasting problem with `coreboot-sdk`:

To compaile coreboot master `convert` is used to support logo in EDKII. `convert` is in `imagemagic` it cannot be added inside coreboot-sdk container because of version confilict with libc, rebuilding coreboot-sdk is also problematic because debian:sid switched to GCC12 and older EDKII branches will not build with that since fixes were introduced later.

@MrChromebox I invited you to [Dasharo Matrix Space](https://matrix.to/#/#dasharo:matrix.org) to chat about contribution and potential synergy.

pietrushnic, 2022-09-04:
What I was able to achieve today is working rebase of `2022.07.08` on `dasharo/master`, which is very close to `coreboot/master`. [Number of workarounds](https://github.com/Dasharo/coreboot/pull/228/files/c504ef90fc0e9dc8473965315f35b0741cbc15af..e447f5482f1a11b41386e33d6fdddbf982988719) I had to applied and conflicts I had to resolve spent couple hours, but it works. Next step is to use Dasharo EDKII to get UEFI Secure Boot and other features. I will probably test that tomorrow.

pietrushnic, 2022-09-05:
To continue story I tried [this](https://github.com/Dasharo/coreboot/tree/dasharo_google_celes/rel_v0.1.0) and bricked my platform, then used [unbricking](https://wiki.mrchromebox.tech/Unbricking) documentation with which I seamlessly recovered. Aparently there is something in @MrChromebox EDKII fork that is really needed for booting my chromebook. Next step is to try `upp_202207` branch and merge to it Dasharo EDKII code.

MrChromebox, 2022-09-05:
@pietrushnic sounds like the best solution is to pull in the Dasharo secureboot patches into my uefipayload_202207 branch (which is based on top of edk2 master commit 2677286307c67b62c198cf6890d62ec540dd8c99). Let's sync up on Matrix

Firminator, 2022-09-25:
You guys rock. I just love the collaboration. 3mdeb and MrChromebox make it happen :100: 
(disclaimer: running a Toshiba Chromebook with MrChromebox FW for several years now, and running a Dell 9010 with Dasharo, and soon the MSI Z690-A with Dasharo).

pietrushnic, 2022-09-25:
@Firminator thank you for your support. We believe we can do even more with such enthusiastic community members as you.

I believe next step here is to rebase Dasharo and push is upstream to EDKII. I'm not sure how realistic is merging things upstream, but we will try it.

pietrushnic, 2022-11-15:
@MrChromebox I'm getting back to this project since I need to create newer version of disposable laptop using CELES. IIUC I should base on 2022.10.24, correct?

MrChromebox, 2022-11-15:
@pietrushnic for coreboot yes, for edk2 my current release is using upp_202210

coolstar, 2023-03-26:
See: https://github.com/coolstar/edk2/tree/upp_202301_secureboot_pxe

for working Secure Boot and PXE

MrChromebox, 2023-03-27:
pulled this into an updated edk2 branch: upp_202304 and will be part of the upcoming coreboot-4.20 based release

MrChromebox, 2023-05-17:
Added in MrChromebox-4.20.0 firmware release

closed 2023-05-17 by MrChromebox

bam80, 2024-04-19:
> Added in MrChromebox-4.20.0 firmware release

Has it happened?

I have this fw but still no evidence of iPXE support:
> **   Device: Asus Chromebox 3 / CN65 (TEEMO)
**   Fw Ver: MrChromebox-4.22.2-5-g822782a1001 (02/13/2024)