MrChromebox/firmware #294: Chromebox CN62 UEFI firmware update success, but blank screen after reboot

onejun0616, 2021-08-14:
Just updated the Asus Chromebox CN62 with latest firmware released 2021.07.25(I had no choice since it was bash command)
After successfully done for update, the device is on but no boot process. Just blank screen(actually monitor does not recognize)
I tried on old monitor, but still no response.
my CN62 is genuine environment at the moment. 4GB Ram and 16GB SSD
Screw on the mainboard was removed, so I put it back, but no response.
I hope there is enough information to be reported as new issue.

Thanks.

MrChromebox, 2021-08-14:
my test CN62 is running the 4.14 firmware just fine, so I doubt there's an issue with the firmware itself - likely just a bad flash. See: https://wiki.mrchromebox.tech/Unbricking

c3r0t1, 2021-08-25:
This exact thing happened to me yesterday as well with the same model and spec with the same firmware. I've ordered a ch341a USB programmer and will give "unbricking" a go to try the flash procedure again if I manage to unbrick my device, fingers crossed.

onejun0616, 2021-09-03:
I finally got ch341a yesterday and followed the instruction under Unbricking page. It's working now. thank you for your detailed instruction.

closed 2021-09-03 by onejun0616

MrChromebox, 2021-09-03:
glad to hear you got it sorted :)

frank-dms, 2021-10-09:
I have the exact same problem. It appears to try and access the USB drive when I reboot but the screen never displays a signal.

lyonelf, 2024-10-28:
Hi,

same issue with cn62

How to prevent that when i'll flash another NC62 ?

MrChromebox, 2024-10-28:
@lyonelf same issue when doing what exactly? Flashing from ChromeOS? Updating from older UEFI firmware? Something else?

MrChromebox, 2024-10-29:
@lyonelf when the firmware flash is complete and you return to the main menu, unplug the device rather than shutting down. The stock firmware is overwriting something during the shutdown procedure I'm guessing

lyonelf, 2024-11-05:
Hi,

Ch341 programmer received and i have error messages.

1st one : couldn't open device 1a86:5512
error programmer initialization failed

If i unplug and replug it i have

spi_send_command: failed to write xx bytes

tested on raspberry and on a laptop

What do i wrong ?

MrChromebox, 2024-11-05:
>  couldn't open device 1a86:5512
> error programmer initialization failed

almost certainly means you have the clip connected backwards.

> tested on raspberry and on a laptop

I don't know what you mean by this

MrChromebox, 2024-11-05:
I would double check that the pins in the blue PCB are inserted into the correct slots on the ch341a. The chip clip wiring to the blue PCB and to the chip appear correct.

lyonelf, 2024-11-05:
yes it is:

[image]

If i connect the clip only with pin 1 to 4, the yellow light wake up.

If i connect the clip with 5 to 8, no.

Mystery

lyonelf, 2024-11-05:
UPDATE:

The ship on the programmer is a ch241B and not A

Is it important ?

MrChromebox, 2024-11-05:
no idea, never heard of a ch341b

muhammad-shoaib-ali, 2024-11-14:
Hello @MrChromebox i just installed firmware on my acer c740 Chromebook and i am facing black screen issue. I searched and found i need to unbrick. But the programmer usb is not available in Pakistan also suzy cable. Can you please help me to fix this issue i really need to solve my email is muhammad.shoaib.mern@gmail.com and whatsapp +923209279820.

Really appreciate it.

MrChromebox, 2024-11-14:
@muhammad-shoaib-ali the c740 can be unbricked with any USB programmer - ch341a, ch347, or raspberry pi. It doesn't support a SuzyQ cable as no USB-C port. But do a hard reset (Refresh+Power) and see if that helps

muhammad-shoaib-ali, 2024-11-14:
Anyother way like if i disassemble ssd storage from the board and install linux in only ssd of Chromebook than assemble again with the board? Or i still need to unbrick first?

MrChromebox, 2024-11-14:
those images are not loading it seems, but your text implies that the device is booting and you are able to select your USB from the boot menu, which means your device is not bricked at all

muhammad-shoaib-ali, 2024-11-14:
Yes i have not removed write protect screw. A video on YouTube about firmware guide he actually did the first of removing screw which i did not remove and stuck on boot menu everytime. It shows Boot manager, use boot image A, B, c and d a.

Thats it. If i remove only screw, it will work again?

rwood56847, 2025-01-22:
> [@lyonelf](https://github.com/lyonelf) when the firmware flash is complete and you return to the main menu, unplug the device rather than shutting down. The stock firmware is overwriting something during the shutdown procedure I'm guessing

This worked after I attempted to flash 2 CN62 boxes with the full UEFI firmware and rebooted using the R + Enter from the firmware util. This 3rd box now boots to UEFI. I will need to unbrick the others with your guide.

@MrChromebox thank you for putting all this together. It is wonderful to be able to use still good hardware for other purposes after Google makes them EOL.

lyonelf, 2025-01-22:
Yep i think this is the problem, reboot and no unplug replug

To unbrick, i have a huge problem, all CH341 that i buy are a B version.

So i have the led ON when i link the clamp.

Is it because it's a voltage issue and i need to follow a tutorial to sold differently the chip ?

MrChromebox, 2025-01-22:
> So i have the led ON when i link the clamp.
> 
> Is it because it's a voltage issue and i need to follow a tutorial to sold differently the chip ?

the ch341a has two LEDs: power, and run. power should be bright red at all times when connected to the host machine. run will be yellow when reading/erasing/writing

lyonelf, 2025-01-22:
Not on 3 differents devices

Yellow stay lightly ON  when the clamp is plugged in to the bios ship

MrChromebox, 2025-01-22:
> Not on 3 differents devices
> 
> Yellow stay lightly ON when the clamp is plugged in to the bios ship

then your wiring is wrong or the clip orientation is wrong.

lyonelf, 2025-01-22:
orientation is good
CH341B voltage issue, maybe ! ?

MrChromebox, 2025-01-22:
nothing to do with voltage - 3.3v vs 5v isn't going to cause the run LED to come on when the programmer isn't active.

something is not connected properly. I have a CN62 here, I've flashed it with a 5v ch341a, no problem at all.

lyonelf, 2025-01-22:
i repeat, impossible to have a CH341A, only B versions.

if you want, i can send you my motherboard via Mondial Relay. Maybe you'll see somthing

MrChromebox, 2025-01-22:
> i repeat, impossible to have a CH341A, only B versions.

I don't know what you mean by this. Are you saying one cannot buy the ch341a anymore, only B?

I have no experience with the -B version, though I suspect that is no the problem here. 

The run light being on means there is power from the programmer being fed back through the data lines, due to mis wiring or configuration. Without seeing your setup, I cannot debug further

lyonelf, 2025-01-23:
The chipset on the CH341 is finished by a B and not a A

[image]

Red light is ok when i connect it on USB port

[image]

Yellow light is a bit ON when i plug the clamp on the Bios

[image]

MrChromebox, 2025-01-23:
sounds like bad hardware then, works fine here (this is a ch3471, but same thing)

[image]

lyonelf, 2025-01-23:
i tried 3 different CH341 and this is the same yellow led bug.

MrChromebox, 2025-01-23:
I'm not sure what to tell you. same wiring each time?

lyonelf, 2025-01-23:
No i changed everytime i bought a new Kit.

So 3 modules, 3 sockets, 3 wire/clamp

Same résult