MrChromebox/firmware #296: No sound in acer chromebox cxi3 (sion)

cesetxeberria, 2021-08-20:
I've done a full rom / uefi replacement. Firmware version is 4.14 (07/25/2021).
Installed windows 10 pro and Debian bullseye. None of them have audio through hdmi or jack/headphones. No audio output devices are found.

MrChromebox, 2021-08-20:
this isn't a firmware issue, it's a driver one. 

Audio jack works under Win10 with the drivers supplied on r/chrultrabook. HDMI audio doesn't work using the built-in jack, but does if using a USB-C to HDMI/DP adapter.

Linux support depends on kernel, config, alsa, SoF, etc.

closed 2021-08-20 by MrChromebox

whitewolf0131, 2024-06-12:
Hi, I installed fedora linux on my acer cxi3 sion. However hdmi audio is not working. Any help please?

MrChromebox, 2024-06-12:
I don't do OS support, that's what the chrultrabook project is for: https://docs.chrultrabook.com