MrChromebox/firmware #298: MSR 0x150 writes blocked, needed for undervolting

FireEgl, 2021-08-30:
MSR 0x150 writes seem to be blocked by the BIOS, is this something you could allow?
It's necessary for undervolting, using https://github.com/georgewhewell/undervolt 

I'd specifically like to use it with the akemi ROM (Lenovo Ideapad Flex 5 Chromebook).

Thanks

rhjdvsgsgks, 2021-10-17:
after some search, i found that "MSR 0x194 - bit[20] is usually what locks MSR 0x150" (http://forum.notebookreview.com/threads/the-throttlestop-guide.531329/page-1280#post-11080969), but i can't set 0x194 bit 20 by `wrsmr` on the fly. could anyone have a try to disable it by set `FLEX_RATIO_LOCK` to 0 in coreboot?

rhjdvsgsgks, 2021-11-04:
is it possible to use https://github.com/psyq321/PowerMonkey as a example?

MrChromebox, 2021-11-04:
maybe, would have to look at it more closely

rhjdvsgsgks, 2022-05-27:
> there's zero chance of me adding microcode that isn't sourced directly from Intel to any of my builds. If people want to try to do it themselves, have at it, just understand that you're going to have a bricked device if you don't supply the correct microcode for your specific CPU (microcode is loaded by CPU ID)

i tried replace cpu_microcode_blob.bin with microcode from SF314-58G (and lot of other device that have sig=0x806ec) which proved able to undervolt (such as https://www.techpowerup.com/forums/threads/intel-i5-10210u-throttling.278487/). but still unable to undervolt. is there anything i missed?

MrChromebox, 2022-05-27:
> is there anything i missed

yes, the microcode in CBFS isn't actually used in most cases. There's a second copy which is written to the FIT (firmware interface tables) and gets read/loaded before coreboot is executed. You'd need to replace that copy too.

rhjdvsgsgks, 2022-05-27:
[this](https://doc.coreboot.org/soc/intel/fit.html) page said "The FIT resides in the BIOS region". so i dump the BIOS region and binwalk in. found that microcode in it has been been modified when using cbfstool downgrade the cpu_microcode_blob.bin file. and system log also showed microcode is the modified version. so i think microcode is already downgrade succeed. but still can't undervolt

MrChromebox, 2022-05-27:
IDK then, sorry

MrChromebox, 2024-07-08:
> can you check that is such of option avaliable in your firmware? [Dasharo/coreboot@92967db](https://github.com/Dasharo/coreboot/commit/92967db1d5f422530bfc2936939cc00c6ab2eee7) [Dasharo/dasharo-issues#159](https://github.com/Dasharo/dasharo-issues/issues/159)

the linked changes are specific to the mainboard in question, there's no generic option to enable them. At best they are platform specific.

closed 2026-11-04 by MrChromebox