MrChromebox/firmware #301: Oses won't load from disk; chromebook brings me back to the chromeos recovery screen. What should I do?

Meganium97, 2021-09-11:
This is the same issue as https://github.com/MrChromebox/scripts/issues/161 but I'm not sure what issue tracker it goes under.

MrChromebox, 2021-09-11:
I'm not exactly sure what you're trying to do or what the problem is. There is no flag to boot an unsigned OS (ChromeOS image), that's what Developer Mode does. If you want to boot Linux, then that's what Legacy Boot Mode / Alternative Bootloader or the UEFI Full ROM firmware is for.

Meganium97, 2021-09-11:
I've figured it out. I installed debian and set up grub and it booted up just fine. I assume its just the difference between how Manjaro and Debian sets up the boot process. I'll close this for now.

closed 2021-09-11 by Meganium97