MrChromebox/firmware #302: likely not feasible Idea? legacy -> UEFI /w ChromeOS compatability

DanHolli123, 2021-09-12:
would it be possible/feasible to replace legacy boot (seabios) with a UEFI firmware so that I can switch back to ChromeOS? (nostalgia reasons, it was my first **personal** computer)

MrChromebox, 2021-09-13:
The RW_LEGACY payload can be any payload, including Tianocore. The problem is that in almost all older Chromebooks, the stock firmware is so buggy that Tianocore will fail to load, and there's no way to get any debugging info out of those older devices.

DanHolli123, 2021-09-19:
@MrChromebox 
> The RW_LEGACY payload can be any payload, including Tianocore.
How could I do this? It seems like it's only possible to overwrite the entire firmware from a look on your site

I do have backup plans if it doesn't work as long as ChromeOS can boot so if it doesn't work I could just go back to SeaBIOS

MrChromebox, 2021-09-19:
> I do have backup plans if it doesn't work as long as ChromeOS can boot so if it doesn't work I could just go back to SeaBIOS

sure, it's always safe to flash RW_LEGACY. What device are you using?

DanHolli123, 2021-09-20:
Acer C720p (peppy)

Sorry if you didn't notice my question as the quote got messed up

How could I install Tianocore in place of SeaBIOS while leaving ChromeOS Verified Boot alone?

MrChromebox, 2021-09-20:
> How could I install Tianocore in place of SeaBIOS while leaving ChromeOS Verified Boot alone?

you need to create a new RW_LEGACY CBFS that contains only Tianocore, then flash it. RW_LEGACY has no impact on ChromeOS whatsoever.

basically, extract the Tianocore payload from the PEPPY full ROM, create a new CBFS, add the payload, then flash it
`cbfstool coreboot_tiano-peppy-mrchromebox_20210725.rom extract -n fallback/payload -f tianocore.fd -m x86`  
`cbfstool rwl_tiano-peppy.bin create -m x86 -s 0x00200000`  
`cbfstool rwl_tiano-peppy.bin add-payload -n payload -f tianocore.fd -c lzma` 
`flashrom -p internal -w -i RW_LEGACY:rwl_tiano-peppy.bin`  

cbfstool is part of the coreboot repo (util/cbfstool) or can be downloaded from my site:
https://mrchromebox.tech/files/util/cbfstool.tar.gz

>Acer C720p (peppy) Haswell

I'd be very skeptical that PEPPY's firmware from 2013 is sane enough for Tianocore to boot.

DanHolli123, 2021-09-20:
awesome, thank you! I'll try to give it a shot tonight

btw, is `coreboot_tiano-peppy-mrchromebox_*.rom` downloaded by or contained in cbfstool?

MrChromebox, 2021-09-20:
it's downloaded by my script from my site
https://mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-peppy-mrchromebox_20210725.rom

DanHolli123, 2021-09-21:
unfortunately line 3 (the one with add-payload) ended with an error saying that it isn't "SELF" despite the first command saying it is
I tried again and got the same error

i continued anyways just out of curiosity and goodbye RW_Legacy XD
time for my backup plan

MrChromebox, 2021-09-21:
> unfortunately line 3 (the one with add-payload) ended with an error saying that it isn't "SELF" despite the first command saying it is. I tried again and got the same error

I tested all these commands here myself. I just uploaded a newer build of cbfstool, so try again using that

> i continued anyways just out of curiosity and goodbye RW_Legacy XD
> time for my backup plan

it's RW_LEGACY, the backup plan is to reflash using my script from ChromeOS

DanHolli123, 2021-09-21:
after some struggling like a dumbass (forgot WHY the script uses the `install` command) and figuring out to use it, finally got `flashrom` to run, successfully flashed, now to see it it's functional

DanHolli123, 2021-09-21:
It switches to Tianocore, But it doesn't seem like it will display (just shows the bitmap removed developer mode warning without automatically booting to ChromeOS like the blank image) when flashed into RW_LEGACY
unfortunately I haven't used Tianocore before so I can't guess and check booting from a drive
I'll give it a second shot with only one drive and see if it works

MrChromebox, 2021-09-21:
I'd try using a different Tianocore payload, since that one is re-initializing the display which may be problematic. Try extracting from https://mrchromebox.tech/files/firmware/full_rom/coreboot_tiano-link-mrchromebox_20210725.rom as that one won't reinit the display.  Otherwise, not much can be done since it's coreboot/depthcharge that are problematic

Better question: why in god's name are you running the stock firmware still on PEPPY? Flash the UEFI firmware and use Brunch to load a newer ChromeOS build if you want to still use ChromeOS, then it will be updated and have more features

DanHolli123, 2021-09-21:
It does work! the graphics do not function and won't automatically use usb, but I switched to my CloudReady drive which in verified boot will disable fs verity and shell, but using Tiano enables fs verity and shell

I'll give that a shot

DanHolli123, 2021-09-21:
> Better question: why in god's name are you running the stock firmware still on PEPPY? Flash the UEFI firmware and use Brunch to load a newer ChromeOS build if you want to still use ChromeOS, then it will be updated and have more features

It was my first computer I owned and I'm extremely sentimental about it. between Chrubuntu, Crouton, Linux Mint, a terrible attempt at windows 10 technical preview (driver issues prevented install), my highly modded MC (I bet you can imagine how slow it was was 400 mods), first taste of Terarria, running android apps through ARC before official support was even rumored and many times I thought I bricked it, it and I have been through a lot

 I've had all this with the PEPPY firmware and official builds, and since the stock 16GB drive isn't useful for much else, I want to keep it for a nostalgia trip no matter what I've done and it won't matter if app support ever gets removed as they're already installed on the drive

DanHolli123, 2021-09-21:
with link it shows... it's a bit glitched, but is functional. Here's a picture if you'd like
https://www.dropbox.com/s/v2d84a301t5y2s2/20210920_204109.jpg?dl=0

every screen displays similarly, but I'm perfectly happy with it as is
Thank you for helping me out with this! it may not be pretty, but it works! I didn't even think it would be possible

DanHolli123, 2021-09-21:
Just a small issue, but added boot entries aren't saving 
Not a major deal I'll just find a way to chainload, but just in case you didn't didn't know

closed 2021-09-21 by DanHolli123

MrChromebox, 2021-09-21:
Depthcharge is initializing the screen to 1024x768 and passing the framebuffer to Tianocore, but apparently that's not happening smoothly. It could probably be debugged, but improving RW_LEGACY on EOL devices isn't where I want to spend my free time =P

Boot order entries can't save, coreboot (Google's 2013 fork thereof) doesn't provide the mechanism for Tianocore to do so