MrChromebox/firmware #303: Morpheus Ryzen GPU unusable

helprequester4394, 2021-09-14:
I had to reinstall my linux and there is a new issue I think with firmware.
On morpheus, there is an issue where trying to load any gpu driver causes X to crash. If amdgpu is blacklisted, and vesa is used then I get an error saying efi can't load the gpu (vesa). If amdgpu is not blacklisted screen is garbled. On previous install amdgpu could be blacklisted and vesa would work.
This may be my kernel but I don't think so since I have been trying different linuxes for one week trying to fix.
This leaves fbdev as the driver but it crashes frequently.

I do not know where I would begin to fix this issue, but I would not mind being pointed to how to fix.

MrChromebox, 2021-09-14:
it's a kernel issue. ChromeOS is stable with the same firmware, right? :)

5.15 should bring some relief, or so I'm told.

FriederHannenheim, 2021-09-21:
You could try using the Linux mainline or GalliumOS kernels.

MrChromebox, 2021-09-21:
the GalliumOS kernel is from year ago and doesn't support hardware from 2017 on, so the chances of it working on MORPHIUS or similar hardware is exactly zero

FriederHannenheim, 2021-09-21:
Oh yeah. Looks like the project is unmaintained. Didn't know that

65a, 2021-11-09:
`iommu=pt` on kernel command line, as well as enabling AMDGPU recovery and disabling memory encryption `mem_encrypt=off` seems to resolve this issue. It appears to be a IOMMU problem with either the hardware or how the kernel (amdgpu and friends) are using it.

closed 2026-11-04 by MrChromebox