MrChromebox/firmware #304: Lenovo C13 Yoga Chromebook. Hard drive is not detected in TianoCore bios

FriederHannenheim, 2021-09-18:
I tried formatting the drive with gpt and msdos partition tables but it didn't show up in the bios. I confirmed that it didn't work by using the map command on the uefi shell

MrChromebox, 2021-09-18:
what type of internal storage does your device have? the NVMe drive shows/boots properly here

FriederHannenheim, 2021-09-19:
On linux it shows up as /dev/mmcblk. After some googling it looks like that's some type of internal so card.

MrChromebox, 2021-09-19:
sounds like Tianocore isn't detecting the eMMC for some reason. I'll need to give you a debug build, have you boot Linux, and pull the debug log. Drop me an email MrChromebox at gmail

FriederHannenheim, 2021-09-19:
Sent it!

MrChromebox, 2024-04-28:
issue resolved in latest MrChromebox release

closed 2024-04-28 by MrChromebox