MrChromebox/firmware #309: Acer CB3-532 - Brasswell no sound

OsirianOne, 2021-11-22:
When installing different flavors of Ubuntu, Raspberry OS, LinuxFX (aka WindowsFx), and Mint.  The sound does not configure, the sound card is not recognized.

I have been rolling this around for more than a year and at one point so frustrated I would execute any snipit of code to try to resolve the issue.

I found today from Intel this page and specifics about the sound and which of the two card methods would be used when modifying a Chromebook and if Linux is directly installed as the primary operating system (not installed on ChromeOS, dual boot, dual booted with Windows)  This seems to indicate an issue with licensing and Intel, when it does not fit one of the 4 acceptable scenarios?  Seems inconceivable you paid for the hardware, and Intel is restricting you from using it.  And even paid for the OS that was on it.....

Is that the case or should I upload my information and what I am experiencing now to help try to resolve the issue.  Some users out there have had just some dumb luck, the kernel and the stars align and it just happens to work or is this a solvable issue?

Section 8 - Chromebooks and SOF
https://thesofproject.github.io/latest/getting_started/intel_debug/introduction.html

MrChromebox, 2021-11-22:
> I have been rolling this around for more than a year and at one point so frustrated I would execute any snipit of code to try to resolve the issue.

IDK what to tell you, it works out of the box in most SOF distros as long as the audio firmware blobs are included. Ubunto 21.04 and Manjaro 21.x.x are both known to work with all Braswell ChromeOS devices. 

But even if not, this wouldn't be a firmware issue, it would be an OS/driver one

closed 2021-11-22 by MrChromebox

OsirianOne, 2021-11-25:
Hi again,
This is my current installation.
Linux <redacted-mdns>  5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
So it might work if I get the distro to update to 21.04; it might possibly work?  I can create a persistent version of 21.04 on a thumb drive and try it boot-able and see?

For the version I have I found the SOF files for the chips, but I think that something is reporting the wrong chip-set or numbers and it doesn't try to pull in the correct ones.

MrChromebox, 2021-11-25:
I'm saying I know it works OOTB on a live USB of Ubuntu 21.04 and Manjaro 21.x.x (with proprietary drivers). I know that the SOF drivers and blobs/firmware are needed. Beyond that, I can't say

OsirianOne, 2021-11-26:
It worked right off the bootable stick, immediately.

So I am going back Standard Ubuntu branch and abandon the unbuntu-mate
branch.

This is certainly resolved now.
Thank you, been a pleasure working with you, and getting straight answer
for once.

I REALLY appreciate the feedback.

MrChromebox, 2021-11-27:
great, glad to hear it :)