MrChromebox/firmware #310: Any Linux distro fails to install on Lenovo Thinkpad 11e: mmc0: timeout waiting for hardware interrupt

Minater247, 2021-11-26:
Hello! Hopefully this is the right place for this, if not please let me know and I'll continue my search elsewhere!

I've been trying for the past few months to get some form of Linux installed on my Chromebook (Lenovo Thinkpad 11e Yoga), but no matter what I try, it always has issues with the integrated emmc. Windows installs perfectly fine, ran it off of the emmc for a couple weeks with no drive issues. However, when I try and install any linux distro, I get this message:

`mmc0: timeout waiting for hardware interrupt`

and a state dump. I can install to the SD card slot (mmc1) without issue - just the integrated one that has problems.

I'm running it off an arch USB at the moment, I will be happy to provide any more info if necessary!

MrChromebox, 2021-11-27:
what board name is shown in the UEFI firmware header?

Minater247, 2021-11-27:
Possibly this is what you're talking about:
`Lenovo Thinkpad 113/Yoga Chromebook  (G3)  ()`
since that's what the UEFI header says, unless I'm mistaken on what you're referring to there.

MrChromebox, 2021-11-27:
it doesn't say ULTIMA inside the parentheses? odd.

I don't know why you'd be getting eMMC errors unless the eMMC is dying. Assuming you're trying to boot a relatively recent distro?

Minater247, 2021-11-28:
Nope, just an empty set of parentheses. Tried doing it in the linux terminal with dmidecode too, but most slots were empty, if that matters.

It works fine on chromeOS, unless it's just hiding errors from me. The drive's only ~4 years old by now, so I don't know how it would be dying - is there any way to check that to be sure, though? All I see are disk/ssd tests.

Also, thought it might be worth a mention that I tried to install neverware cloudready today it it was throwing errors about a bad GPT partition table, but never gave me the emmc timeout message.

MrChromebox, 2021-11-28:
sounds like something is awry. I don't know how the device name is correct but the board name missing, given the device name is derived from the board name, and not the other way around.

I'd recommend reverting to the stock firmware and seeing if ChromeOS is fine at this point

Minater247, 2021-11-28:
Since I've had this since 2017, I don't believe it's a morphius. Going off of [this chart](https://www.chromium.org/chromium-os/developer-information-for-chrome-os-devices) it's about 2 years too early, I got this in 2018.

Also, just since you had said so initially, I've done this a couple of times and both times just given up on it and gone back to chromeOS. No trouble there. I have an old photo of the boot screen from an earlier try and it had the same thing, empty parentheses.

MrChromebox, 2021-11-28:
when running my script from ChromeOS, what board name is reported? I wonder if the string is simply too long and so the insertion of the board name fails...

Minater247, 2021-11-28:
Running the script in galliumos live image to reinstall chromeOS, it says` ULTIMA`. Is that enough, or should I completely reinstall chromeOS to run the script from there?

MrChromebox, 2021-11-28:
no, that's enough - it tells me it's just a display bug and not something more serious.

this is the first time anyone has reported an issue with ULTIMA's eMMC. I'll see if there's any tweaking to be done on the firmware side. Does the issue occur when using the stock firmware + RW_LEGACY to boot Linux?

Minater247, 2021-11-28:
It works perfectly on Legacy! Absolutely no issues, gallium install running off of eMMC. I'll hold off installing all my arch stuff until testing is done, so you can do whatever you need to.

chusx, 2022-07-15:
having this exact issue on ULTIMA with the latest full uefi firmware. windows 10 installs and works fine, but linux distros on kernel 5+ fails to install with this error  "mmc0: Timeout waiting for hardware cmd interrupt." (ubuntu 22.04, 20.04, debian bullseye)

chusx, 2022-07-15:
there is a [debian bug](https://bugzilla.kernel.org/show_bug.cgi?id=73241) with similar symptoms, it seems like `aspm` handling between the bios and os might be an issue. the suggested fix `debug_quirks=0x40 debug_quirks2=0x4` did not work on ULTIMA unfortunately.

mmccall0813, 2022-10-23:
any updates on this? I am having the same issue on the latest full ROM firmware

MrChromebox, 2022-10-24:
I'm pretty sure this isn't a firmware issue - either a Linux kernel or a hardware one

gridbugs, 2023-03-27:
I just got a 11e chromebook and I get the same problem. When I booted into chromeos it reported the board name as "ULTIMA" but in the coreboot settings the heading is "Lenovo Thinkpad 11e/Yoga Chromebook  (G3)  ()" (empty parens). I've tried installing alpine linux and during installation it freezes (I think while it's copying files to the disk) and dmesg contains "mmc0: timeout waiting for hardware interrupt".

MrChromebox, 2023-03-27:
> coreboot settings the heading is "Lenovo Thinkpad 11e/Yoga Chromebook (G3) ()" (empty parens).

UEFI settings, not coreboot. And 'ULTIMA' isn't shown because the string is too long for the field, not a big deal.

without having one of these on hand, going to be very time consuming to try and figure out what's wrong with the eMMC on some of them with the UEFI Full ROM firmware

Yorper, 2023-05-24:
@MrChromebox I’ve got a few of these 11e 3rd gens being retired from fleet. Happy to powerwash one and donate it to you?

Yorper, 2023-05-24:
> @Yorper sounds good, drop me an email at mrchromebox at gmail

Email sent 👍

Yorper, 2023-05-24:
@MrChromebox email sent 👍

superatrain, 2023-10-31:
Given this bug, it would be nice to add the legacy firmware option back to the firmware scripts, as the coreboot option is not really usable today.

kquote03, 2023-11-17:
> there is a [debian bug](https://bugzilla.kernel.org/show_bug.cgi?id=73241) with similar symptoms, it seems like `aspm` handling between the bios and os might be an issue. the suggested fix `debug_quirks=0x40 debug_quirks2=0x4` did not work on ULTIMA unfortunately.

This actually worked for me for some reason...

MArabie, 2024-01-05:
I would like to resurrect this issue. I have done a little digging and have a lead.
TLDNR:
The kernel is looking for " (dmi_match(DMI_BIOS_VENDOR, "LENOVO") ". My theory is our machines have the Google code names
[image]
GOOGLE != LENOVO and so the timeout occurs.
I see there is/was a Set Hardware ID function in the MrChromebox script, but I do not see it in the 4.22 version (or maybe its only available to some machines). Then again maybe its not that easy of a fix.

Research:
From this post someone found using an older Ubuntu 18.04 (not 18.04.5) allowed successful install. (I have confirmed). Freezing the kernel allowed update further.
From this post there was a similar issue found for Ubuntu. (https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1848883). And a hacky kernel patch was applied. This has made it into the mainline Linux. (https://github.com/torvalds/linux/commit/afd7f30886b0b445a4240a99020458a9772f2b89)

MrChromebox, 2024-01-05:
In theory if I changed the DMI vendor to LENOVO it should match, but looking at the patch, it's only going to match on the Geminilake eMMC controller, not Braswell, since they have different PCI IDs. What I should do is see if it can be disabled in the firmware init

MArabie, 2024-01-05:
Good catch. I missed that GLK is Geminilake.
Probably means the workaround kernel patch did not capture the true issue. Or the kernel parch needs an additional OR in the evaluation. Wonder if it affects Appolo Lake Lenovos as well (between Braswell and Gemini)

MrChromebox, 2024-01-05:
AFAIK, Ultima is the only device with the issue, or at least no one else has reported a problem.

This really does seem like a hardware + kernel driver issue, and I don't see a good way to work around it in firmware -- the capabilities registers are read only.

skajmer, 2024-01-05:
> Research: From this post someone found using an older Ubuntu 18.04 (not 18.04.5) allowed successful install. (I have confirmed). Freezing the kernel allowed update further. (https://www.reddit.com/r/chrultrabook/comments/13iydpr/mmc0_timeout_waiting_for_hardware_interrupt/?sort=new) From this post there was a similar issue found for Ubuntu. (https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1848883). And a hacky kernel patch was applied. This has made it into the mainline Linux. ([torvalds/linux@afd7f30](https://github.com/torvalds/linux/commit/afd7f30886b0b445a4240a99020458a9772f2b89))

I made that comment about 18.04 on that post (I own the SrebrnyBrek64 account), this was pretty hacky and took like an entire day to do the upgrades.

HighOnVoltage, 2024-01-29:
So I can also confirm a custom version of windose 10 works on this device; However because of storage constraints you'll always run into issues. 

Also I have noticed the SD card gets corrupted for extra storage, I believe there's a bug in the coreboot firmware IMO. Didn't some give MrChromebox one of these to play around with? 

At the end of the day I was just trying to save the device from being landfill.

skajmer, 2024-01-31:
> Also I have noticed the SD card gets corrupted for extra storage
SD Card seems to work fine on my machine, no corruption there. The only thing that happens it sometimes isn't mounted automatically on boot but that's about it.

skajmer, 2024-02-16:
I had a bit of time, I heard Linux works on rw_legacy so I decided to test it. And it seems to work. I had edit the script because you can't use rw_legacy on eol devices. I know it's unsupported, but still it does work.
[image]

zvirja, 2024-02-18:
@skajmer Could you please clarify how to use your fix? Is it something you have to change in OS configs, or what? Sorry, a newbie here 😅

skajmer, 2024-02-18:
@zvirja 
I wouldn't exactly recommend it for a newbie but, that's how you do it:
1. You go back to the stock firmware (Using the option in the firmware utility script)
2. Then you need to fork the  firmware utility script and edit it to bypass EOL check so you can flash rw_legacy
3. Flash rw_legacy with your edited script
4. After flashing rw_legacy you can press Ctrl+L on the White dev mode enabled screen thing to go to boot menu
5. After pressing Ctrl + L SeaBios will start press ESC to select a device to boot from
6. Boot to install media and install desired distro as normal (After installation you'll need to run chromebook audio script because i've found a glitch with it that this script fixes)

jincopunk, 2024-06-25:
I also ran into this issue, is it worth removing the ✅ from UFEI column or adding a note in the linux column for Lenovo ThinkPad 11e/Yoga Chromebook (G3)?

MArabie, 2024-10-04:
@skajmer, I see you forked repo is gone. Is there a private link or did you nuke it? Thanks.

skajmer, 2024-10-04:
@MArabie I nuked it since I don't even have a chromebook anymore.

wand-am, 2026-09-09:
ran into this issue just yesterday while trying to install arch on a used 11e g3. the fixes outlined in #585 (passing `sdhci.debug_quirks=0x60` to the kernel) worked well for me. i tried the DD fix described above but it did not unfortunately help in my case (unless something went wrong with the zeroing, but of course it's not something i can check now). curious to see if this issue is figured out eventually :)

MrChromebox, 2026-09-09:
I've implemented proper edk2 support for eMMC/SD devices in ACPI mode, and switching Ultima's eMMC/SD back to ACPI mode seems to have resolved the issue - please verify:

[coreboot_edk2-ultima-mrchromebox_20260909.zip](https://github.com/user-attachments/files/32014366/coreboot_edk2-ultima-mrchromebox_20260909.zip)

skajmer, 2026-09-13:
I might verify this when I have time, I unbricked what's left of my ultima few months ago

MrChromebox, 2026-09-13:
well, I'm fairly confident in this solution as the two ultimas I have here now work properly, so I'm going to close this after the 2609 release unless someone has a good reason otherwise

skajmer, 2026-09-13:
I tested it, seems to work fine. You can close this.

zvirja, 2026-09-13:
@MrChromebox Wow, amazing job!

Could you point to the patch place? I am curious what you did and how you handled it 😊

MrChromebox, 2026-09-13:
> [@MrChromebox](https://github.com/MrChromebox) Wow, amazing job!
> 
> Could you point to the patch place? I am curious what you did and how you handled it 😊

1) add new driver to edk2 which supports eMMC in non-PCI mode (ACPI like BYT/BSW, MMIO like AMD)
2) add new cbmem table entry to pass non-PCI eMMC/SD to edk2
3) select it for BYT/BSW/AMD PCO
4) switch ultima eMMC/SD from PCI mode back to ACPI mode (like ChromeOS uses). It was switched to PCI in upstream coreboot because there was no support in edk2 for ACPI mode devices at the time.

*1) https://github.com/MrChromebox/edk2/commit/5342f428abac83167ce191559a36bca29f20a06c
*2) https://review.coreboot.org/c/coreboot/+/94054
*3) https://review.coreboot.org/c/coreboot/+/94055
*4) https://review.coreboot.org/c/coreboot/+/95392

refutationalist, 2026-09-24:
This is great!

I'm away from my chromebooks, but will test as soon as I'm able.

MrChromebox, 2026-09-26:
fixed in MrChromebox-2609 release

closed 2026-09-26 by MrChromebox