MrChromebox/firmware #312: disable auto start opening lid Lumpy, howto modify firmware file

dogedivenexia, 2021-12-05:
good afternoon,
brilliant work, thanks a lot.
as from subject I was looking how to midify the firmware file to disable the function making the LUMPY chromebook start automatically as the lid is open.
regards

dave

MrChromebox, 2021-12-07:
in file src/mainboard/samsung/lumpy/smihandler.c, you could try adding `case EC_LID_OPEN:` in between lines 19 and 20, then recompile

Emulatore, 2023-01-28:
Thank you MrChromebox for you work, and sorry if I aggregate to this request, if cause disorder i can move to a new request, I am having the same problem on my asus c436fa aka Helios, is possible to add option in coreboot firmware for enable or disable booting from opening lid and plugging in charger? thank you in advance for your reply

mikepapa64, 2025-04-14:
there is no src in file system, it is just sbin sys

closed 2026-11-04 by MrChromebox