MrChromebox/firmware #313: Dratini HP Laptop no sound

mjaxn2057, 2021-12-07:
The onboard sound card is not recognized by any linux distro.

MrChromebox, 2021-12-07:
that's... not a firmware issue. Use a kernel which includes SOF support and the SOF-blobs (or install them manually). Manjaro 21.x (with proprietary drivers) and Ubuntu 21.04 should both work. But again, not a firmware issue.

closed 2021-12-07 by MrChromebox

mjaxn2057, 2021-12-07:
I am using a kernel with SOF support (Fedora 35), and I have set the kernel module parameters to use the SOF module, but it still seems that the audio card is not detected by the kernel. I will try with Manjaro to see if that works. Thanks.

MrChromebox, 2021-12-07:
maybe you need to install the sof-blobs package? Either way, best troubleshooted on the appropriate Fedora forums/support venue