MrChromebox/firmware #314: HP Chrombook x360 14c shows us as HP Pro c640 Chromebook (DRATINI)

mjaxn2057, 2021-12-11:
HP Chrombook x360 14c shows us as "HP Pro c640 Chromebook (DRATINI)" when using the firmware utility script. 

Is it still safe to use the utility?

MrChromebox, 2021-12-11:
is your device DRATINI? There are often multiple models with the same board name. The board name is all that matters

mjaxn2057, 2021-12-13:
If you go by your Supported Devices list, I think my device is a DRAGONAIR. How do I check to see what my board name is?

MrChromebox, 2021-12-13:
boot to recovery mode or developer mode, or check the HWID via chrome://system

gjskha, 2021-12-20:
I have a HP Chromebook 14c and the HWID is ELDRID

MrChromebox, 2021-12-21:
why are you opening multiple issues for the same thing?

DmitriyBabaev, 2022-01-11:
Hi, I have the same Chromebook X360 14c and my son installed the custom bios and didn't make a backup. I tried 
to install Ubuntu and Win10, but it looks like not all devices work, especially touch screen and audio. Now I want to rollback to Chrome OS. Would you be able to help me with that? Do you have any original firmware for the Chromebook?

mjaxn2057, 2022-01-12:
Unfortunately, I lost the backup I made of my Chromebook. If you do find a copy, please let me know.

closed 2022-01-12 by MrChromebox