MrChromebox/firmware #316: Lenovo 14e firmware issue with the RW_LEGACY

ssemakin, 2021-12-13:
Hi, first of all big thanks for the good job on the MrChromebox.tech project and all the scripts!

The specific issue in question is to enable easy USB booting (into Linux) with my Lenovo 14e (LIARA) chromebook. 

Is it possible to launch pre-installed SeaBIOS from the U-Boot shell?

What I did was the following. I install the `RW_LEGACY` using `mrchromebox.tech/firmware-util.sh` script, I get the successfully installed status. I reboot, press CTRL+L and get into the white screen and the text "Alternative firmware menu" and that was pretty much the dead end. 

I managed to remove the Write-Protect, updated firmware using the `chromeos-firmwareupdate --mode=recovery` command. When pressing the CTRL+L, I get now a menu with 2 options: 1) U-Boot bootloader 2) TianoCore bootloader. Choosing the TianoCore just gives a beep. U-boot gets into the bootloader shell. Running the `firmware-util.sh` script doesn't change anything. 

The "Supported Devices" page at mrchromebox.tech has the note for this chromebook: _AMD: Some StoneyRidge devices (CAREENA? BARLA?) may not currently have functional Legacy Boot Mode due to bugs in Google's firmware._ So I'm assuming this is the reason `firmware-util.sh` can't do anything. 

So I thought, as a workaround would it be possible at all to start SeaBIOS from the U-Boot shell? Otherwise I guess nothing can fix it other than modifying Google firmware itself?

MrChromebox, 2021-12-13:
> What I did was the following. I install the RW_LEGACY using mrchromebox.tech/firmware-util.sh script, I get the successfully installed status. I reboot, press CTRL+L and get into the white screen and the text "Alternative firmware menu" and that was pretty much the dead end.

well it should show the menu with just Tianocore; you can try blinding booting by pressing 1 or 2 and see if that works

>I managed to remove the Write-Protect, updated firmware using the chromeos-firmwareupdate --mode=recovery command.

just to restore the stock RW_LEGACY firmware?

> When pressing the CTRL+L, I get now a menu with 2 options: 1) U-Boot bootloader 2) TianoCore bootloader. Choosing the TianoCore just gives a beep. U-boot gets into the bootloader shell. Running the firmware-util.sh script doesn't change anything.

A working U-boot and a broken Tianocore is correct for the stock firmware on AMD Stoney devices

> So I thought, as a workaround would it be possible at all to start SeaBIOS from the U-Boot shell? Otherwise I guess nothing can fix it other than modifying Google firmware itself?

why would you do that, vs simply adding SeaBIOS as a menu item?

Also, if one could launch a bootable SeaBIOS payload from the alternative bootloader menu, I would be providing that as an option

ssemakin, 2021-12-14:
> well it should show the menu with just Tianocore; you can try blinding booting by pressing 1 or 2 and see if that works

You are totally right! Basically that fixes this ticket :D So I got another laptop with the Write-Protect off and the original firmware, I just get into the "Alternative Firmware menu" white screen, and both 1 and 2 buttons work fine. 

I'm getting into another challenge to bypass the Secure boot and boot from the self-signed kernel (Ubuntu works fine, but non-Ubuntu needs to be danced with). If there's any tips on that, would be much helpful! :)

> just to restore the stock RW_LEGACY firmware?

Yup, I read that I don't need to remove Write-Protect or do anything to the firmware, but then I read the note about buggy Google firmware and getting into the white screen without any options made me think that maybe there's updated firmware from  Google with the fix. So I ran the command recommended by google:

```
Usually for people who wants to “update all my firmware to right states”, do:
      
    chromeos-firmwareupdate --mode=recovery
```
So I guess I'll need to restore the firmware to the factory one on that chromebook.

> A working U-boot and a broken Tianocore is correct for the stock firmware on AMD Stoney devices

I guess it's easier to attach a screen, I'm not sure I'm using correct terms. This is what firmware looks like on the laptop where I didn't "update" it:

[image]

And on the laptop where I "updated" it (where the Tianocore is broken) all 4 sections show the 139 version.

> why would you do that, vs simply adding SeaBIOS as a menu item?

> Also, if one could launch a bootable SeaBIOS payload from the alternative bootloader menu, I would be providing that as an option

Ok, I see. My experience with chromebooks is limited to somewhat 3-5 days, I don't fully understand how this all works with its firmware so probably I simply don't make much sense. I assumed that SeaBIOS is installed on some of the MMC disk partitions and when we get into U-Boot, we can "boot" into SeaBIOS so to speak ( because Tioanocore wasn't working). But as I said I probably don't make much sense 🤷

ssemakin, 2021-12-14:
Sorry, and of course if you'd like me to run some tools and collect debug data on one of those laptops, I can definitely do that.

MrChromebox, 2021-12-14:
> Ok, I see. My experience with chromebooks is limited to somewhat 3-5 days, I don't fully understand how this all works with its firmware so probably I simply don't make much sense. I assumed that [...]

Chromebooks have a region/slice in the firmware image named RW_LEGACY, which contains the menu list and payload(s) to run. They are located in flash, not on disk, and are loaded into memory/executed by the ChromeOS payload (depthcharge). 

Tianocore should boot on LIARA (at least it does on mine) but something is screwy in the latest depthcharge with the menu text color/rendering. That's on google, not me.

closed 2021-12-17 by ssemakin