MrChromebox/firmware #319: Dell Latitude 5400 Chrome (google/sarien): GNU/Linux: Super key not working and caps lock acts as super key

paulmenzel, 2021-12-20:
[Split out from issue https://github.com/MrChromebox/firmware/issues/318.]

Installing RW_LEGACY (TianoCore) using the [Chrome OS Firmware Utility script](https://github.com/MrChromebox/scripts) worked fine.

In Ubuntu 21.10 or Debian sid/unstable, the super key (globe icon) does not work, and the caps lock key seems to be mapped to the super key.

paulmenzel, 2021-12-20:
When pressing key X, Linux logs:

Super key:

    [  371.749464] atkbd serio0: Unknown key pressed (translated set 2, code 0xd4 on isa0060/serio0).
    [  371.749472] atkbd serio0: Use 'setkeycodes e054 <keycode>' to make it known.
    [  371.848416] atkbd serio0: Unknown key released (translated set 2, code 0xd4 on isa0060/serio0).
    [  371.848430] atkbd serio0: Use 'setkeycodes e054 <keycode>' to make it known.

Rectangle function key (F3):

    [  392.224626] atkbd serio0: Unknown key pressed (translated set 2, code 0xd5 on isa0060/serio0).
    [  392.224642] atkbd serio0: Use 'setkeycodes e055 <keycode>' to make it known.
    [  392.366896] atkbd serio0: Unknown key released (translated set 2, code 0xd5 on isa0060/serio0).
    [  392.366910] atkbd serio0: Use 'setkeycodes e055 <keycode>' to make it known.

Overview function key (F4):

    [  426.163166] atkbd serio0: Unknown key pressed (translated set 2, code 0xd6 on isa0060/serio0).
    [  426.163183] atkbd serio0: Use 'setkeycodes e056 <keycode>' to make it known.
    [  426.246926] atkbd serio0: Unknown key released (translated set 2, code 0xd6 on isa0060/serio0).
    [  426.246942] atkbd serio0: Use 'setkeycodes e056 <keycode>' to make it known.

paulmenzel, 2021-12-20:
`libinput debug-events` detects the “Caps Lock” key as `KEY_LEFTMETA (125)`.

coolstar, 2023-06-16:
Not a coreboot issue. Key mappings are controlled by Wilco EC which is closed source

closed 2023-06-17 by paulmenzel