MrChromebox/firmware #320: Dell Latitude 5400 Chrome (google/sarien): GNU/Linux: Touchscreen does not work

paulmenzel, 2021-12-20:
[Split out from issue https://github.com/MrChromebox/firmware/issues/318.]

Installing RW_LEGACY (TianoCore) using the [Chrome OS Firmware Utility script](https://github.com/MrChromebox/scripts) worked fine.

In Ubuntu 21.10 and Debian sid/unstable, the touchscreen is not working. Linux reports:

    [    0.000000] microcode: microcode updated early to revision 0xea, date = 2021-01-05
    [    0.000000] Linux version 5.13.0-19-generic (buildd@lgw01-amd64-013) (gcc (Ubuntu 11.2.0-7ubuntu2) 11.2.0, GNU ld (GNU Binutils for Ubuntu) 2.37) #19-Ubuntu SMP Thu Oct 7 21:58:00 UTC 2021 (Ubuntu 5.13.0-19.19-generic 5.13.14)
    […]
    [    0.000000] DMI: Dell Inc. Sarien/Sarien, BIOS Google_Sarien.12200.155.0 07/29/2020
    […]
    [    9.352368] i2c_designware i2c_designware.0: controller timed out
    [    9.352386] mip4_ts i2c-MLFS0000:00: nothing at this address

sayad23, 2024-12-04:
Hello dad I have just an issue with the formatting my laptop for me to have Linux on it 
And I am using Mr Chromebox.tech script 
I arrived on the latest steps to make it work but I have as you can see in the picture the option of install and update you UEFI firmware grey 

If anyone know anything on how we can make it available to do it please let me know.

[image]

MrChromebox, 2024-12-04:
> In Ubuntu 21.10 and Debian sid/unstable, the touchscreen is not working. Linux reports:

nothing I can do about that, it's an issue with the stock firmware not the RW_LEGACY bootloader

>  the option of install and update you UEFI firmware grey

correct, it is not available for this model

closed 2024-12-04 by MrChromebox

VGerris, 2025-09-26:
update: it did work at installation but now there is issues loading the driver on latest 24.04 LTS updates. there is some error about acpi not being able to set power change, so it's a driver bug or issues coming from firmware. It may work on an older kernel.