MrChromebox/firmware #323: Booting from 'USB Device' failed;verify it contains a 64-bit UEFI OS

Zewwy, 2021-12-28:
Long story short. I got a CN60 ChromeBox. I wanted more than what was on it, which was already Kodi based on the setup-kodi.sh

I read everything on the site, and learnt how it all worked, and managed to update the seaBIOS by grabbing the latested copy of the setup-kodi,sh script and running it under the 'chronos' account on the ChromeOS alternative console. Everything seemed fine and I was able to boot Ubuntu Desktop, and even a Windows installer ISO. but the drive listed was heavily partition, which was to be expected with the whole ChromeOS/Kodi Dual boot thing goin' on.

So I got back into ChromeOS [Ctrl+D], got in the alt console as chronos, ran the setup-kodi.sh and selected to install the custom UEFI firmware. Now I can see the rabbit icon at boot, but no matter how I created my ISO images, or use my IODD device I just keep getting the same error:

"Booting from 'USB Device' failed;verify it contains a 64-bit UEFI OS"

I don't get it.... I tried multiple USB sticks, multiple valid x64 installer images, multiple USB creation apps (Rufus, Etcher)... what happened?

Zewwy, 2021-12-28:
Huh... I pressed ESC, at boot to get the UEFI menu, MrChromeBox-4.14. Went into Boot menu, found my IODD Device with Ubuntu installer ISO mounted, and it booted right up... Sweet, Thanks for this firmware. Not sure why the default boot order wasn't picking up on it.

Zewwy, 2021-12-28:
Everything seems to be working perfectly, closing this issue, Thanks again for this amazing firmware, and helping people out.

closed 2021-12-28 by Zewwy

UNTUNGKSR, 2024-01-22:
Hello, I have the same issue here. Lenovo Chromebook C340-15 360, i3-8130 w /126eMMC. Detect as Yoga C630 Pantheon device in mrchromebox rec.
After following all the tutorials from mrchromebox w newest firmware 4.22.1, on the 1st try, I succeeded in booting the Windows Inst method, but I couldn't install it because there was not enough space. And then I cleaned up all the Chromebook partitions w/out a backup of it partition or the recovery. Then I rebooted my Chromebook, and this issue appeared, it can't recognize 64bit UEFI OS. Tried with Rufus and Ventoy but still, it doesn't work. Restart again to boot from the USB drive, from the boot file, nothing. But if I tried that USB on a Windows Machine, it worked normally both of them successfully booted without any problems.
So, please if any solution for this issue. Thx a lot mate.

onbronem, 2025-04-05:
Hey I’m getting a message on my hp 14 saying “booting from usb: general flash disk, failed verify it contains a 64 bit UEFI OS” do you know how I can resvole the issue