MrChromebox/firmware #325: Celes support

tejasraman, 2022-01-02:
My grandma has a CELES Chromebook (Samsung Chromebook 3) nearing its EOL. I plan to try and install Brunch on it (we are considering getting her a new laptop) using this UEFI firmware. 

She likes to use Duolingo, and I wonder if the linux-chromebook 5.4 kernel has support for the internal mic of ALC5650(chtrt5650) Braswell devices like Celes?

FYI, I posted this on Brunch too

MrChromebox, 2022-01-02:
sorry, what exactly is the firmware issue you're having/wondering about? I have no idea about kernel driver support. It should work with an upstream kernel 5.14+ with sof and sof-blobs

tejasraman, 2022-01-02:
I am wondering about Brunch on UEFI Full ROM firmware. I am also condsidering Linux. If Brunch is not a good option (only supports up to 5.10) I will use Linux instead.

Brunch has a ChromeBook kernel (Seems like other Braswell devices were tested working with it) but no one mentioned the mic. 

Thanks!

tejasraman, 2022-01-02:
Here’s a snippet from Brunch FAQ: 
 Several kernels can be enabled throught the configuration menu: 

one of them is called linux-ChromeBook(which says it provides better support for Chromebooks on your UEFI firmware)

Thanks for the advice about the SOF blobs though!

tejasraman, 2022-01-02:
And how exactly do I get the SOF blobs on Ubuntu 21.10?

MrChromebox, 2022-01-02:
The best place for Brunch support is their discord channel. AFAIK, there are no issues with my firmware preventing Brunch from running properly.

> And how exactly do I get the SOF blobs on Ubuntu 21.10?

I don't know offhand, but if I had to guess, `sudo apt install sof-blobs` or similar.

closed 2022-01-02 by MrChromebox

tejasraman, 2022-01-02:
Googled it - SOF on Ubuntu is `sudo apt install firmware-sof-signed`