MrChromebox/firmware #330: Add "CONFIG_USE_OPTION_TABLE=y" in .config

wzn556, 2022-02-06:
Hello, Thanks for your work for Chromebook.
I'd like to know is it possible to add "CONFIG_USE_OPTION_TABLE=y" in ".config_" files to make bios setting(like VT-d , Intel ME, etc.) can be changed via nvramtool or coreboot configurator.
I've tried to add it by myself, but there are always many mistakes. I 'll appreciate it if you can add it in your official files. This will provide us with many conveniences.
Thanks.

MrChromebox, 2022-02-06:
I would need to not only add that config option for each board, but create the CMOS defaults for each, which is a lot of work when we're talking about 100+ boards, and most users don't need to change the default configuration

MrChromebox, 2022-08-24:
closing as will not implement

closed 2022-08-24 by MrChromebox