MrChromebox/firmware #331: ASUS Chromebox 4 does not detect microSD for boot

theflyingape, 2022-02-13:
FYI, I can use a USB adapter, but not the native microSD slot:
```*********************************************************
**   Device: ASUS Chromebox 4 (DUFFY)
** Platform: Intel CometLake
**  Fw Type: Stock ChromeOS w/RW_LEGACY
**   Fw Ver: Google_Duffy.13324.71.0 (09/03/2020)
**    Fw WP: Enabled
*********************************************************```

MrChromebox, 2022-02-13:
did you try removing/re-inserting the micro-SD card, then going back into the boot menu?

theflyingape, 2022-02-14:
Greetings, I now tried that eject/insert to no change detected. Also, the UEFI shell map before/after SD card insertion does not show anything differently, and exiting back to boot menu shows me only the NVMe internal storage.

I can boot off a Fedora Live 32gb USB stick with the bootable 64gb microSD card from a working ASUS Chromebox **3** (which works perfectly) into this 4 box:
```
$ df -Tht ext4
Filesystem          Type  Size  Used Avail Use% Mounted on
/dev/mapper/live-rw ext4  7.3G  5.4G  2.0G  74% /
/dev/mmcblk0p1      ext4  966M  271M  630M  31% /run/media/liveuser/BOOT
/dev/mmcblk0p2      ext4   58G   53G  1.8G  97% /run/media/liveuser/FEDORA

$ dmesg | grep mmc
[    3.283741] mmc0: SDHCI controller on PCI [0000:00:14.5] using ADMA 64-bit
[    3.284115] mmc1: CQHCI version 5.10
[    3.284294] mmc1: SDHCI controller on PCI [0000:00:1a.0] using ADMA 64-bit
[   78.505803] mmc0: new ultra high speed SDR104 SDXC card at address aaaa
[   78.522220] mmcblk0: mmc0:aaaa SH64G 59.5 GiB 
[   78.528940]  mmcblk0: p1 p2 

$ lspci | grep SD
00:14.5 SD Host controller: Intel Corporation Comet Lake PCH-LP SCS3
00:1a.0 SD Host controller: Intel Corporation Device 02c4
02:00.0 Non-Volatile memory controller: Sandisk Corp WD Blue SN550 NVMe SSD (rev 01)

$ sudo fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 59.48 GiB, 63864569856 bytes, 124735488 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x7e2c6bf3

Device         Boot   Start       End   Sectors  Size Id Type
/dev/mmcblk0p1 *       2048   2050047   2048000 1000M 83 Linux
/dev/mmcblk0p2      2050048 124735487 122685440 58.5G 83 Linux
```

MrChromebox, 2022-02-15:
I'm not able to reproduce this on my CML Chromebox (CTL model) - inserting a micro-SD card then entering the boot menu, the card is detected and bootble. I'll have to see if anything different on the Asus model, but I suspect not. Guessing it might just be a firmware incompatibility with that particular SD card on that device - esp since it works on your CB3

theflyingape, 2022-02-15:
Agreed, thanks for weighing in on this. This CB4 splash startup menu is definitely different (not the typical ugly white requiring Ctrl/D) from all the rest of the Chrome devices I've dealt with. I have the alternate in USB booting for now to do my tinkering.

closed 2022-02-15 by theflyingape

MrChromebox, 2022-02-15:
yes, 10th-gen Chromeboxes and newer now use this nice dark-themed developer mode screen with the alternative bootloader menu. I need to update the documentation on my site for this