MrChromebox/firmware #332: HP CB1 recognized as ASUS CN60 on/after update

guidol70, 2022-02-22:
I have a HP Chromebox with a SeaBIOS Full ROM Legacy
from 2016 which I updated via the firmware-script (in a live-Linux session under Linux Mint)

I had as only option Full ROM UEFI udate and my HP CB1 has been recognized as a ASUS CN60 (maybe because its updated to 8GB Ram and 128GB SSD?)

In the UEFI firmware it now shows ASUS CN60 (CPU seems to be the same)

Its running - since some days - "Chrome OS Flex" (CloudReady 2.0) - also with the ASUS CN60 UEFI Firmware.

Is there a chance to get the right Model-Name back?

MrChromebox, 2022-02-22:
back in 2016 there was only one firmware image for all the Haswell Chromeboxes, now (since a few years ago) each device gets its own proper image. You'll need to restore the stock firmware for Zako, then flash the UEFI firmware back:

* boot Linux USB (can't flash from CrOS Flex)
* run script
* restore stock firmware
* choose N for have backup
* select Zako / HP G1 when prompted
* do not reboot yet
* flash RW_LEGACY firmware
* set GBB flags to Legacy / 2s
* reboot
* boot Linux USB again
* flash UEFI firmware
* reboot back into CrOS Flex

guidol70, 2022-02-23:
> * select Zako / HP G1 when prompted
> * set GBB flags to Legacy / 2s

selected Zako / HP CB1 when prompted
set GBB flags to Legacy / 1s

Many Thanks - worked flawless ;)
[image]

closed 2022-02-23 by guidol70