MrChromebox/firmware #334: No sound ASUS Chromebook Flip

Sirmadsen, 2022-03-02:
Hi,

First of all, thanks for the work you are doing. The best and probably the easiest way I've found.

Anyhow, to the problem. Everything is working flawlessly and installation is a breeze. The only thing I can't get to work is sound. I've tried all Ubuntu flavors and different version with no luck. Even tried installing repositories and following several guides. I'm thinking that it might be a firmware issue?

My Chromebook is still in R/W and I only installed the RW LEGACY firmware. The information I can get about the sound device is that it is 'Intel Corporation Sunrise Point-H HD Audio', if that helps.

Sorry if I'm unclear, I'm new at the whole Linux thing and just wanted to make my unused Chromebook into a Linux machine.

Best regards

MrChromebox, 2022-03-02:
this isn't a system firmware issue, it's a complex mix of kernel/driver/alsa/dsp firmware. I'm not aware of any way to get audio working on SKL/KBL devices under Linux when using a recent kernel

Sirmadsen, 2022-03-02:
Ah, so it's just waiting for someone to come up with a fix or just live with no sound?

MrChromebox, 2022-03-02:
use BT or a USB adapter. Any fix would require someone very knowledgeable with Linux audio to forward-port the necessary CrOS audio components into a format usable by mainline linux

Sirmadsen, 2022-03-02:
Bummer... Thanks a lot xD

dtb43, 2022-03-15:
Was chasing this one down in a linux Mint forum just this week, and came to the same end result.  HP Dratini laptop (HP chromebook x360 14c) with MrChromebox firmware (full rom) with sound not working (not the firmware's fault!).  Had one of the USB-C to 3.5mm adapters around, and it works.  Will try BT speakers as well.

Side note, thanks for the great work, Mr. Chromebox.  Kids' laptops got really unstable under ChromeOS recently, followed measures from your .tech website, firmware works great, Linux installs are a cinch.  Keep up the great work if you can...it's much appreciated!  Now begins the search for solving touchscreen functionality (less optimistic).

closed 2022-03-15 by MrChromebox

theyazhouse, 2025-04-12:
Hello
I have tried Ubuntu, MX Linux and Linux Mint no sound but Manjaro works if anyone is looking for a linux distro to try. It worked straight out of the box. I have an Acer Chromebook so you know.