MrChromebox/firmware #335: Procedure to override EC RW firmware

bam80, 2022-03-07:
Original firmware had a feature to "Preventing the RW EC firmware from being overwritten by Software Sync at boot" via GBB (Google Binary Block) flags:
https://review.coreboot.org/plugins/gitiles/chrome-ec/+/fd9d07de6d6f688f80e489d67977ad1070a08a15/README.md#Preventing-the-RW-EC-firmware-from-being-overwritten-by-Software-Sync-at-boot

Now we don't have GBB (but Software Sync was implemented), so the procedure to override EC RW fw without rebuilding the whole UEFI firmware would be helpful.
https://github.com/MrChromebox/coreboot/commit/f6f02eef9f33103e9d101e2be33edaac54e30e93#commitcomment-68158449:
> while I could make it easier, rebuilding isn't needed - you can just remove the ecrw file and hash from cbfs and flash back

I think instructions how to do that here or on the Wiki would be useful.

MrChromebox, 2022-03-07:
this "feature" may not do what you want - all it would do is prevent **future** EC RW firmware updates from being applied. It wouldn't keep the EC in RO firmware. And if you're sophisticated enough to be building/flashing your own EC RW updates, then disabling EC swsync by removing the update files from CBFS isn't going to be problematic

bam80, 2022-03-07:
Maybe I just need to learn how to work with CBFS, I'm completely unaware of it.
Still some mentioning of it in the docs would be useful, as well as swsync feature itself, as its not what upstream coreboot has I suppose.

closed 2026-11-04 by MrChromebox